PixlOne/logiops #198: Button for gesture scrolling/dragging

davorvr, 2021-01-30:
Hello, thank you for your work on this utility, it's awesome and I use it daily.

It seems that actions may only be keypresses, but I was wondering if there was a way to map a button (specifically the thumb button) via a gesture to a scrolling action? That is, to make a button press activate the gesture mode which then, instead of sending a keypress, performs a scrolling action. This way, when the button is pressed, moving the mouse up or down would scroll vertically, and moving it sideways would scroll horizontally.

Thanks.
Best regards,
Davor

PixlOne, 2021-01-30:
[You can map gestures to an axis](https://github.com/PixlOne/logiops/wiki/Configuration#axis). For an example, here's my config for scrolling up/down with the up/down gesture and moving back/forward with the left/right gestures.

```
{
    cid: 0xc3;
    action =
    {
        type: "Gestures";
        gestures: (
            {
                direction: "Up";
                mode: "Axis";
                axis: "REL_WHEEL";
                axis_multiplier: 0.6;
            },
            {
                direction: "Down";
                mode: "Axis";
                axis: "REL_WHEEL";
                axis_multiplier: -0.6;
            },
            {
                direction: "Left";
                mode: "OnRelease";
                action =
                {
                    type: "Keypress";
                    keys: ["BTN_SIDE"];
                };
            },
            {
                direction: "Right";
                mode: "OnRelease";
                action =
                {
                    type: "Keypress";
                    keys: ["BTN_EXTRA"];
                };
            },
            {
                direction: "None"
                mode: "OnRelease"
                action =
                {
                    type: "None";
                    keys: ["BTN_MIDDLE"];
                };
            }
        );
    };
}
```

davorvr, 2021-01-31:
Nice, that example really helped, works like a charm now. Thank you! Might be worth adding something like this to the example configuration.

closed 2021-01-31 by davorvr