PixlOne/logiops #201: Linux "windows" (super) button code

Piranit, 2021-02-04:
Hello!
I wanted to assign a "windows" (super) button to the gesture button on ubuntu 20.04.
Please tell me which code of the button?
Thanks!

hasimsait, 2021-02-05:
KEY_LEFTMETA. Read the wiki. "For a list of key/button strings, refer to linux/input-event-codes.h." https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h

olehyaremko, 2021-02-08:
@hasimsait thank you so much!
I saw this document.
I was looking for a button with names like "Super", "Windows" or "Special".
Nowhere did I find information that this button is called "Meta".
There was no comment for "KEY_LEFTMETA" in that document.
Now I will know, have a nice day!

Piranit, 2021-02-08:
@hasimsait thank you so much!
I saw this document.
I was looking for a button with names like "Super", "Windows" or "Special".
Nowhere did I find information that this button is called "Meta".
There was no comment for "KEY_LEFTMETA" in that document.
Now I will know, have a nice day!

closed 2021-02-08 by Piranit

dieggsy, 2021-10-06:
To add a bit to the confusion, Alt is sometimes called meta. Good this is at least documented in an issue though.

ElectricRCAircraftGuy, 2023-06-17:
@hasimsait , is it documented anywhere that "meta" means the <kbd>Super</kbd> or <kbd>Windows</kbd> key? Or, did you just identify this by guessing and/or trial and error while looking in the `input-event-codes.h` file?