PixlOne/logiops #202: MX Master 2s Now shows under a different name

Qwerty-Space, 2021-02-04:
My config stopped working, I've got it somewhat working again by changing the "name" attribute to `Wireless Mouse MWireless Mo`.  But my extra buttons still don't work

<details><summary>Config:</summary>
<p>

```cfg

devices: (
{
    name : "Wireless Mouse MWireless Mo"
    smartshift:
    {
        on: true;
        threshold: 99;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1600;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_UP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_W"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "CycleDPI";
                            dpis: [600, 1000];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type = "ToggleSmartshift";
                        }
                    },
                    {
                        direction: "None"
                        mode: "OnRelease";
                        action = 
                        {
                            type: "Keypress";
                            keys: ["KEY_ESC"]
                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "ToggleSmartShift";
            };
        }
    );
}
);

```
</p>
</details>

Qwerty-Space, 2021-02-04:
And after the mouse woke up from sleep the name changed back.  ¯\_(ツ)_/¯

I'm going to leave it open, because perhaps it's worth investigating

Qwerty-Space, 2021-02-06:
And it's back to being `Wireless Mouse MWireless Mo`.  But this time restarting the mouse doesn't fix it

loretanr, 2021-02-14:
Just bought the master 3 and I'm having the same issue. Most of the time it's called 
"Wireless Mouse MX Master 3"  sometimes
"Wireless Mouse MWireless Mo".

Similar basic config, nothing fancy
Running  Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux

Qwerty-Space, 2021-02-15:
What fixed it for me was installing the release version, instead of using the git version

loretanr, 2021-02-17:
Interesting, I just briefly tested the release version which does not work at all for me. But the newest git version works well (as long as I don't make the mistake of starting multiple instances again). So that's a good sign

PixlOne, 2021-02-17:
This bug is caused by how HID++ 2.0 names are sent. logid requests 16 characters at a time (since that is the max that can be requested through the HID++ 2.0 protocol), so it sends "Wireless Mouse M" and then it should send "X Master 3\0" on the second request.

However, due to some bug, it seems that the reports sometimes get sent wrong, or the wrong response is sent to the function call. What I could do as a temporary workaround is request the device name twice and ensure that tit is the same each time. Although, I will look into a more permanent and thorough solution.