PixlOne/logiops #204: Logid fails to start on boot, when restarted after login works flawless

supermar1010, 2021-02-15:
Hello,
first thanks for the great software!

Now my problem is I'm running arch and using the logiops git package from the aur, I recently updated so I'm on the newest version.
Problem is when logid starts while booting it fails to apply the config, `systemctl status logid` shows this:
```
logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; enabled; vendor preset: disabled)
     Active: active (running) since Mon 2021-02-15 16:58:14 CET; 4min 34s ago
   Main PID: 582 (logid)
      Tasks: 6 (limit: 38430)
     Memory: 1.8M
        CPU: 8ms
     CGroup: /system.slice/logid.service
             └─582 /usr/bin/logid

Feb 15 16:58:14 mario-desktop systemd[1]: Started Logitech Configuration Daemon.
Feb 15 16:58:17 mario-desktop logid[582]: [WARN] Error adding device /dev/hidraw2: std::exception
Feb 15 16:58:17 mario-desktop logid[582]: [WARN] Error adding device /dev/hidraw1: std::exception
Feb 15 16:58:17 mario-desktop logid[582]: [WARN] Error adding device /dev/hidraw0: Invalid sub ID
Feb 15 16:58:18 mario-desktop logid[582]: [WARN] Error adding device /dev/hidraw2: open failed: No such file or directory
Feb 15 16:58:18 mario-desktop logid[582]: [WARN] Error adding device /dev/hidraw1: open failed: No such file or directory
Feb 15 16:58:20 mario-desktop logid[582]: [WARN] Error adding device /dev/hidraw4: std::exception
Feb 15 16:58:22 mario-desktop logid[582]: [WARN] Error adding device /dev/hidraw3: std::exception
Feb 15 16:58:22 mario-desktop logid[582]: [WARN] Error adding device /dev/hidraw0: Invalid sub ID
```
So it seems like it is trying to start but either failing to add the device or failing to see my unifying receiver.

After restarting my status looks likes this
```
logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; enabled; vendor preset: disabled)
     Active: active (running) since Mon 2021-02-15 17:08:35 CET; 13s ago
   Main PID: 5095 (logid)
      Tasks: 7 (limit: 38430)
     Memory: 764.0K
        CPU: 16ms
     CGroup: /system.slice/logid.service
             └─5095 /usr/bin/logid

Feb 15 17:08:35 mario-desktop systemd[1]: Started Logitech Configuration Daemon.
Feb 15 17:08:37 mario-desktop logid[5095]: [WARN] Error adding device /dev/hidraw4: std::exception
Feb 15 17:08:38 mario-desktop logid[5095]: [WARN] Error adding device /dev/hidraw3: std::exception
```

Anghellos, 2021-03-03:
Same problem here. if i run sudo systemctl start logid after boot is working ok.
Any suggestion?

ZzTriplett, 2021-03-05:
Same issue here. I was using my mouse over Bluetooth and it seemed to be working, but now I am using a unify receiver and it only works after a second run.

TillDeathCoder, 2021-03-11:
> Same issue here. I was using my mouse over Bluetooth and it seemed to be working, but now I am using a unify receiver and it only works after a second run.

I have the same problem when started to use the unifying device. No problems with Bluetooth only

Cherkah, 2021-03-27:
same problem here too (... std::exception)
systemctl enable logid is set
occurs on unifying pc & bluetooth pc.

my logid.cfg :
```
devices: ({
2   │ name: "Wireless Mouse MX Master 3";
3   │ buttons: (
4   │ { cid: 0xc3;
5   │   action: { type: "ChangeHost"; host: "previous"; }
6   │ });
7   │ });

```

Manjaro + KDE

ReillyBrogan, 2021-03-30:
I've noticed similar behavior when my mouse has gone to sleep due to inactivity. When I move it again and it wakes up the mouse defaults to default settings until I restart logid. I wonder if it's the same issue.

Aditeya, 2021-03-31:
I've has the same issue at first, then i tried using `sudo systemctl enable logid` and that seems to have fixed the problem.

ReillyBrogan, 2021-04-01:
@Aditeya All that command does is flag the service to be started on boot. If that fixes it for you then your issue was just that you forgot to do that in the first place (most services do not default to enabled when they are installed).

The rest of us are experiencing a different issue.

ReillyBrogan, 2021-04-04:
You know I just noticed this starting working perfectly for me again, and it got me wondering if this is something library related. My system is Solus which just updated a few packages, perhaps the latest Systemd fixes it? We went from 247.4 to 247.6.

Aditeya, 2021-04-04:
I'm also getting the issue where if the mouse is left untouched it goes to powersave mode, and when I turn the computer on it doesn't apply the config. But if i move the mouse, which activates it, and then i turn it on the computer the config is applied.

ReillyBrogan, 2021-04-06:
Yes, I wonder if my experience with it starting to work was just that I was touching the mouse before I started the computer.

IE, the root cause of this issue is that logid is seemingly not handling a mouse coming online.

jixbo, 2021-04-08:
I'm facing a similar issue on Ubuntu 21.04, using the latest logiops.
I'm not getting a std::exception tho, I'm getting "Invalid function ID" when booting the laptop.
`systemd[1]: Started Logitech Configuration Daemon.
logid[1893]: [WARN] Error adding device /dev/hidraw1: Invalid function ID
logid[1893]: [WARN] Error adding device /dev/hidraw1: Invalid function ID
`
After restarting the service, it works fine.

kris7t, 2021-04-08:
I was experiencing similar issues under Arch linux with systemd 248rc4, but it went away after upgrading to systemd 248.

It seems the difference is that now the kernel module `hid-logitech-hidpp.ko` gets incorporated into my initramfs, so that the Logitech driver is initialized in early userspace.

jixbo, 2021-04-17:
Seems like @kris7t was right. I have fixed the issue by adding hid-logitech-hidpp.ko to the initframfs.
For debian based distros, like ubuntu, the solution is to add "hid-logitech-hidpp" to /etc/initramfs-tools/modules file, then run `sudo update-initramfs -u`.

ReillyBrogan, 2021-04-17:
Hmmm I'm not sure that's the correct solution in all cases. I am using Solus which pre-bakes the initramfs into the kernel package and that initramfs already contains the `hid-logitech-hidpp` module. It also doesn't explain why the mouse goes back to an unconfigured state when it wakes back up after being unused for a while.

I'm thinking there might be two separate issues at play here:
- Missing `hid-logitech-hidpp` module in the initramfs
- Logid not handling "wake" events from the mouse

I suspect that if the second issue was solved it would probably result in the first fix no longer being needed as well.

jixbo, 2021-04-19:
I have tested not touching the mouse until the the system has booted, and I did not have any issue. There might be two separate issues.

thiagolucio, 2021-05-02:
I solve this problem in Fedora 33 just made install use dnf "sudo dnf install logiops" and when restarting the problem it's solved. Need to wait a feel seconds (I'm don't count).
https://bodhi.fedoraproject.org/updates/?search=logiops

thiagolucio, 2021-05-03:
I believe I have discovered the problem. It does not run automatically when initializing because the "vendor" class of "systemctl" logid.service" has a "disable" status. When this value is disabled it does not start automatically, indicating that you need to do it manually. this status for vendor enabled, I'm still researching.
I found in Gnome (Fedora 33/34), the SystemdManager extension that allows you to manually add a service at startup or at least facilitate the manual process if necessary.
[image]
[image]

ReillyBrogan, 2021-05-05:
@thiagolucio All that means is that the logid service shouldn't be automatically enabled when the package is installed, which I think is the better default. After all somebody installing it is going to need to create a config file anyway so there's no point in having it run until then.

thiagolucio, 2021-05-05:
> @thiagolucio Tudo isso significa que o serviço logid não deve ser habilitado automaticamente quando o pacote é instalado, o que eu acho que é o melhor padrão. Afinal, alguém que o instala vai precisar criar um arquivo de configuração de qualquer maneira, então não há por que executá-lo até então.

But I have the configuration file implemented incorrect place path. It's really a problem I think. But your opinion it's relevant because I don't think in this perspective. I created an alias in my bash to activate the service typing in terminal the command "mx" but I think in the past this service runs automatically right?

ReillyBrogan, 2021-05-07:
@thiagolucio  If you ran `sudo systemctl enable logid` then the service should start on boot (it does for me). Are you experiencing something different after running that command?

thiagolucio, 2021-05-07:
> @thiagolucio If you run `sudo systemctl enable logid` then the service should start on boot (it does for me). Are you experiencing something different after running that command?

Yes. I actually activated the service using this command. I actually researched a lot about it. If you look at the image of the service I posted earlier, you will see that the service is "enable". However, for not being able to force the "enable" of "vendor" in systemctl, it does not start automatically. If you look at the systemctl documentation you will see that "vendor" must be enabled for it to automatically start with the system. It is this problem that I am trying to solve. Anyone else experiencing this problem could do this same check by typing in the bash 'systemctl list-unit-files --state = enabled' and doing this check.

Actually, in my view, this is the service problem that is occurring in this ticket.

ReillyBrogan, 2021-05-08:
@thiagolucio I did read the documentation. The vendor flag determines whether or not the service automatically starts after the service file is installed. However if a user uses `systemctl enable` or `systemctl disable` it overrides the vendor flag. So in your screenshot above where you show that the logid service is state `enabled` and vendor `disabled` then the service starts on boot. Changing the vendor flag doesn't change this behavior in any way.

thiagolucio, 2021-05-08:
> @thiagolucio I did read the documentation. The vendor flag determines whether or not the service automatically starts after the service file is installed. However if a user uses `systemctl enable` [...]

But that is exactly the problem that I noticed. As much as I apply the command "systemctl enable" the status of "vendor" does not change for the logid. It continues to disable. Why I don't know how to tell you. I just put this information in because I thought it might help in some way.

Nicryc, 2021-05-13:
Any news on that ? 
I've tried every solution proposed in this thread and none of them worked for me. Still have this issue, have to restart the service each time the mouse is disconnected.

I'm using the latest release 0.2.3 on Fedora 34 with Gnome 40.

ReillyBrogan, 2021-05-14:
@Nicryc I don't believe anyone has figured it out yet. Unfortunately I'm not familiar enough with how HID works under Linux to properly know how to debug this and it doesn't seem that the dev has had the time/inclination lately to look at it (which is cool, this is a volunteer project after all). 

If anyone has any idea of where to start feel free to let me know (breakpoints in the code? USB protocol capture?)

thiagolucio, 2021-05-14:
I believe that the way Systemd works have changed in relation to the new changes made to it. Especially in Fedora34, which had many new implementations. Here it is working well in the following situations:
- After the system loads it is restarted manually using the command `sudo systemctl restart logid`

- The screen is locked and with that, the mouse turns off / disconnects, and when logging in again unlocking the screen, the logid works.

- Using a gnome-shell extension called "Systemd manager" where you after installing configure to start the service together with the system.

I still believe that if someone finds (because I haven't had time or discovered it yet) a way to permanently enable "vendor" for "enable" status, everything will work as expected.

I fully agree that this is an open project and that everyone has the power to help the developer who just deserves our thanks for so much dedication. In my opinion, we will continue to try and exchange ideas here, one hour we found the solution.

wooparadog, 2021-05-16:
I think it's related to systemd dependency settings:

In the [Install] section of service unit file provided:
```
[Install]
WantedBy=multi-user.target
```
It will make logid start before `multi-user.target` is reached, which might be after some key target(like basic.target). So there might be a chance devices haven't been initialized yet. You can check it by:
```
$ systemctl list-dependencies multi-user.target
```

But I didn't bother to find exactly which target it depends on, so I made the following changes:

1. Edit service file.
    ```
    $ sudo -E systemctl edit logid.service
    ```
2. Add the following content(Change `WantedBy` to graphical.target)
    ```
    [Unit]
    After=multi-user.target
    Wants=multi-user.target
    
    [Install]
    WantedBy=graphical.target

    ```
3. Reload daemon and reenable logid
    ```
    $ sudo systemctl daemon-reload
    $ sudo systemctl enable logid --now
    ```

I've done some reboots myself, it works every time. But I think more tests are needed to create a fix PR for this issue(or find exactly which target it should start after).

ReillyBrogan, 2021-05-16:
> I've done some reboots myself, it works every time. But I think more tests are needed to create a fix PR for this issue(or find exactly which target it should start after).

I've rebooted a few times with your changes and it's worked for me. I haven't yet tested to see if it works with the mouse going to sleep but it does work with turning off the mouse and turning it back on. It did take ~20 seconds after logging in before it started working but that might be a separate issue.

Taking a look at the dependencies of `multi-user.target` it seems to me that the most likely actual dependency is one of the `systemd-udevd` ones.

wooparadog, 2021-05-16:
@ReillyBrogan Thanks for testing. `graphical.target` is all I need, I'll create a PR in this format then,

thiagolucio, 2021-05-16:
> I think it's related to systemd dependency settings:
> 
> In the [Install] section of service unit file provided:
> 
> ```
> [Install]
> WantedBy=multi-user.target
> ```
> 
> It will make logid start before `multi-user.target` is reached, which might be before some key targets(like basic.target). So there might be a chance that devices haven't been initialized yet. You can check it by:
> 
> ```
> $ systemctl list-dependencies multi-user.target
> ```
> 
> But I didn't bother to find exactly which target it depends on, so I made the following changes:
> 
> 1. Edit service file.
>    ```
>    $ sudo -E systemctl edit logid.service
>    ```
> 2. Add the following content(Change `WantedBy` to graphical.target)
>    ```
>    [Unit]
>    After=multi-user.target
>    Wants=multi-user.target
>    
>    [Install]
>    WantedBy=graphical.target
>    ```
> 3. Reload daemon and reenable logid
>    ```
>    $ sudo systemctl daemon-reload
>    $ sudo systemctl enable logid --now
>    ```
> 
> I've done some reboots myself, it works every time. But I think more tests are needed to create a fix PR for this issue(or find exactly which target it should start after).

Unfortunately for me, nothing changes. The same problem persists.
 systemctl list-dependencies multi-user.target
multi-user.target
● ├─abrt-journal-core.service
● ├─abrt-oops.service
○ ├─abrt-vmcore.service
● ├─abrt-xorg.service
● ├─abrtd.service
● ├─atd.service
● ├─auditd.service
● ├─avahi-daemon.service
● ├─chronyd.service
● ├─crond.service
● ├─cups.path
● ├─firewalld.service
○ ├─flatpak-add-fedora-repos.service
● ├─libvirtd.service
● ├─livesys-late.service
● ├─livesys.service
● ├─logid.service
● ├─mcelog.service
○ ├─mdmonitor.service
● ├─ModemManager.service
● ├─NetworkManager.service
● ├─plymouth-quit-wait.service
○ ├─plymouth-quit.service
● ├─sssd.service
● ├─systemd-ask-password-wall.path
● ├─systemd-homed.service
● ├─systemd-logind.service
● ├─systemd-oomd.service
● ├─systemd-resolved.service
○ ├─systemd-update-utmp-runlevel.service
● ├─systemd-user-sessions.service
● ├─thermald.service
○ ├─vboxservice.service
○ ├─vmtoolsd.service
● ├─basic.target
● │ ├─-.mount
● │ ├─low-memory-monitor.service
○ │ ├─rpmdb-rebuild.service
● │ ├─tmp.mount
● │ ├─paths.target
● │ ├─slices.target
● │ │ ├─-.slice
● │ │ └─system.slice
● │ ├─sockets.target
● │ │ ├─avahi-daemon.socket
● │ │ ├─cups.socket
● │ │ ├─dbus.socket
● │ │ ├─dm-event.socket
● │ │ ├─iscsid.socket
● │ │ ├─iscsiuio.socket
● │ │ ├─libvirtd-ro.socket
● │ │ ├─libvirtd.socket
● │ │ ├─pcscd.socket
● │ │ ├─sssd-kcm.socket
● │ │ ├─systemd-coredump.socket
● │ │ ├─systemd-initctl.socket
● │ │ ├─systemd-journald-audit.socket
● │ │ ├─systemd-journald-dev-log.socket
● │ │ ├─systemd-journald.socket
● │ │ ├─systemd-udevd-control.socket
● │ │ ├─systemd-udevd-kernel.socket
● │ │ ├─systemd-userdbd.socket
● │ │ ├─virtlockd.socket
● │ │ └─virtlogd.socket
● │ ├─sysinit.target
● │ │ ├─dev-hugepages.mount
● │ │ ├─dev-mqueue.mount
● │ │ ├─dracut-shutdown.service
● │ │ ├─import-state.service
○ │ │ ├─iscsi-onboot.service
● │ │ ├─kmod-static-nodes.service
○ │ │ ├─ldconfig.service
● │ │ ├─lvm2-lvmpolld.socket
● │ │ ├─lvm2-monitor.service
● │ │ ├─plymouth-read-write.service
● │ │ ├─plymouth-start.service
● │ │ ├─proc-sys-fs-binfmt_misc.automount
○ │ │ ├─selinux-autorelabel-mark.service
● │ │ ├─sys-fs-fuse-connections.mount
● │ │ ├─sys-kernel-config.mount
● │ │ ├─sys-kernel-debug.mount
● │ │ ├─sys-kernel-tracing.mount
○ │ │ ├─systemd-ask-password-console.path
○ │ │ ├─systemd-binfmt.service
○ │ │ ├─systemd-boot-system-token.service
○ │ │ ├─systemd-firstboot.service
○ │ │ ├─systemd-hwdb-update.service
○ │ │ ├─systemd-journal-catalog-update.service
● │ │ ├─systemd-journal-flush.service
● │ │ ├─systemd-journald.service
○ │ │ ├─systemd-machine-id-commit.service
● │ │ ├─systemd-modules-load.service
● │ │ ├─systemd-random-seed.service
○ │ │ ├─systemd-repart.service
● │ │ ├─systemd-sysctl.service
○ │ │ ├─systemd-sysusers.service
● │ │ ├─systemd-tmpfiles-setup-dev.service
● │ │ ├─systemd-tmpfiles-setup.service
● │ │ ├─systemd-udev-trigger.service
● │ │ ├─systemd-udevd.service
○ │ │ ├─systemd-update-done.service
● │ │ ├─systemd-update-utmp.service
● │ │ ├─cryptsetup.target
● │ │ ├─local-fs.target
● │ │ │ ├─-.mount
● │ │ │ ├─boot-efi.mount
● │ │ │ ├─media-dados.mount
○ │ │ │ ├─ostree-remount.service
● │ │ │ ├─systemd-remount-fs.service
● │ │ │ └─tmp.mount
● │ │ ├─swap.target
● │ │ │ └─dev-zram0.swap
● │ │ └─veritysetup.target
● │ └─timers.target
● │   ├─dnf-makecache.timer
● │   ├─fstrim.timer
● │   ├─logrotate.timer
● │   ├─mlocate-updatedb.timer
● │   ├─raid-check.timer
● │   ├─systemd-tmpfiles-clean.timer
● │   └─unbound-anchor.timer
● ├─getty.target
○ │ └─getty@tty1.service
● ├─nfs-client.target
○ │ ├─auth-rpcgss-module.service
● │ ├─rpc-statd-notify.service
● │ └─remote-fs-pre.target
● └─remote-fs.target
○   ├─iscsi.service
○   ├─var-lib-machines.mount
●   └─nfs-client.target
○     ├─auth-rpcgss-module.service
●     ├─rpc-statd-notify.service
●     └─remote-fs-pre.target
lines 84-135/135 (END)

ReillyBrogan, 2021-05-16:
@thiagolucio I noticed when I set this up that SystemD wasn't smart about removing the old "wants" symlink when I modified the service file and re-enabled it. Check to see if you still have the following file on your system: `/etc/systemd/system/multi-user.target.wants/logid.service`. If so delete it (it's just a symlink), run `sudo systemctl daemon-reload` and `sudo systemctl enable logid`.

ReillyBrogan, 2021-05-16:
@wooparadog I experimented with this a bit more and got it working with `Wants=basic.target, `After=basic.target`, and `WantedBy=graphical.target`. I still see some errors/exceptions in the logid service output however I'm no longer experiencing that it takes ~20 seconds after logging in before logid begins working with my mouse. 

I think that might be a better compared to `multi-user.target` assuming it works for you as well. I think the errors are due to a race between systemd-udevd setting up the hidraw devices under /dev/ and the kernel mapping them to USB devices and can probably be ignored.

thiagolucio, 2021-05-16:
> @wooparadog I experimented with this a bit more and got it working with `Wants=basic.target, `After=basic.target`, and `WantedBy=graphical.target`. I still see some errors/exceptions in the logid [...]

I believe my system has some wrong config or something like that. Just manually to start work. Back to the research from me.

ReillyBrogan, 2021-05-19:
@wooparadog I was looking at the system journal and I noticed that systemd-logind also processes keyboard events. That service depends on `user.slice` and `modprobe@drm.service`. Experimentally I used those for `Wants` and `After` and the last few boots have had logid working as soon as I logged in. I'm not sure if those are the correct ones but it might be worth testing to see if it works for you too. It might be best to try to find whatever is the actual service dependencies just so we can have logid be ready sooner than later.

thiagolucio, 2021-05-20:
I changed my operating system. I am now using Ubuntu LTS 20.04 and performed the installation. It is now operating normally as expected.
I believe the problem really was with my previous system. Fedora 34, Gnome 40.
In addition, in the new installation, the service is working as expected also in "vendor".
[image]
[image]

ReillyBrogan, 2021-05-21:
You know I've half wondered if this might be related to GNOME 40. I'm running that as well.

thiagolucio, 2021-05-21:
> You know I've half wondered if this might be related to GNOME 40. I'm running that as well.

Yes. I thought the same. However, this may also be a recurrence of Fedora 34 itself. In this new version, there have been many changes in the system itself. Assuming btrfs by default, Wayland as a graphical manager, pipeware to manage the sound. Since installing Fedora 34, I have bitterly regretted having removed Fedora33 that I used before because it was extremely stable, while Fedora 34 went through a lot of problems, mainly with Wayland because it really is not yet stable as it should be to be a standard of any system. I believe that all changes in the Fedora 34 ecosystem for these new features in addition to Gnome 40 itself are causing problems with logiops.
In fact, I consider this problem solved because if it is necessary to solve the automatic activation problem in all distributions and all versions, it will never be done again. For those wishing to be premature adopter, they will have to be satisfied with experiencing this type of problem. In my case, as I use my notebook to work I found it more recommended to install an LTS, stable, and Debian distro again and the problem was solved.

brunobmello25, 2021-05-25:
might be related to gnome 40 after all. I'm running arch and faced the same issue.

[image]

bulletmark, 2021-05-26:
As a note to people who come here with this problem, the suggestion above is to change the systemd `[Install]` from `WantedBy=multi-user.target` to `WantedBy=graphical.target`, etc as described above. I am trialing that change myself now to fix this issue, but I should point out to others that it is not enough to simple re-enable the service after changing that. You must first disable the service and then re-enable it, e.g:

````
$ sudo systemctl daemon-reload
$ sudo systemctl disable --now logid
$ sudo systemctl enable --now logid
````

The reason is that the `disable` ensures that the old systemd "wants" symlinks are cleared out (it clears them all regardless of which were set) so it fixes the issue @ReillyBrogan [mentioned](https://github.com/PixlOne/logiops/issues/204#issuecomment-841861932).

closed 2021-05-27 by PixlOne

bulletmark, 2021-05-30:
Maybe it is a separate issue but after a resume from suspend I have to restart logid to make it apply my settings.

wooparadog, 2021-06-16:
Notes for more debugging info:  If you're still encountering this problem. You can check if `logid` is started at last stages of startup, especially after various udev jobs(search `dev`) by following command:

```bash
$ systemd-analyze plot > systemd.svg
```

There might be differences between all kinds of setups.  You can change your systemd target accordingly.

ReillyBrogan, 2021-06-22:
Interestingly a few times I've noticed that logiops will be working when I log in and will stop working a few seconds in. I wonder if we're seeing Gnome break it somehow?

thiagolucio, 2021-06-22:
> Interestingly a few times I've noticed that logiops will be working when I log in and will stop working a few seconds in. I wonder if we're seeing Gnome break it somehow?

Man. I suggest you install "Stacer" and put them to run manually (including vendor boolean to true). 
https://github.com/oguzhaninan/Stacer

I think then solve this (and other's) problems you have around.

Freder211, 2022-01-09:
I tried every solution on this thread, but couldn't manage to make it work. Anyone still having troubles with this issue?

distro: arch linux
wm: qtile

thiagolucio, 2022-01-09:
> I tried every solution on this thread, but couldn't manage to make it work. Is anyone still having trouble with this issue?
> 
> distro: Arch Linux wm: qtile

I don´t use Arch. But in this case, I think using the same solution including in Manjaro it's possible. Another solution you can try is installing Solaar in your system to have a redundancy solution. 

https://archlinux.org/packages/community/any/solaar/
https://github.com/pwr-Solaar/Solaar

wooparadog, 2022-04-15:
:(  

My solution stopped working recently, and it seems that a simple wait for udev is not good enough anymore.. I'm trying more solutions right now.  If anyone is interested, you could help test if this works.

Basically, it's letting udev tagging input devices with systemd, and systemd can then involve those input devices into its dependency handling.

1. udev tagging: make a new rule for udev at `/etc/udev/rules.d/99-input-tagging.rules`
    ```
    ACTION=="add", KERNEL=="event*", SUBSYSTEM=="input", TAG+="systemd", , ENV{SYSTEMD_ALIAS}+="/sys/subsystem/input/devices/$env{ID_SERIAL}"
    ```

2. Reboot
3. Search for logitech related devices with `systemctl list-units --type device| grep -i logi`
4. Add that device to logid.service's `unit` section with `systemctl edit logid.service`
    ```
    After=sys-subsystem-input-devices-Logitech_USB_Receiver.device
    Wants=sys-subsystem-input-devices-Logitech_USB_Receiver.device
    ```
5. Reboot again to see if it works.

thiagolucio, 2022-09-04:
Too much time after this issue was created. This problem does not exist anymore.

brunobmello25, 2022-09-05:
@thiagolucio this issue is still happening today. I'm always having to restart logid after the system boots

thiagolucio, 2022-09-06:
> @thiagolucio this issue is still happening today. I'm always having to restart logid after the system boots

Sorry about that. In my case I don't use logiops anymore. I'm using Solaar. Works great no my Fedora.

thiagolucio, 2022-09-06:
> > > @thiagolucio this issue is still happening today. I'm always having to restart logid after the system boots
> > 
> > 
> > Sorry about that. In my case I don't use logiops anymore. I'm using Solaar. Works great no my Fedora.
> 
> @thiagolucio, are you using it on Wayland? I used to use Solaar on Ubuntu 20.04 LTS, but after the upgrade to 22.04 LTS it stopped working, so I migrate to Logiops.

I'm using the last version of Fedora (36). Fedora has used Wayland for a long time. Pipeware too. No problem at all.

valerionew, 2022-11-03:
This would probably be fixed out of the box for everyone if a new version that includes the fix was released.

@PixlOne is there any particular reason why there is no release yet that includes 5947cc939e?

PR #246 (2021-05-27): Fix systemd denepdency
Let logiops start after multi-user.target is reached, and change it as a dependency to `graphical.target`. Fix #204
commit 5947cc93 (2021-05-27): Fix systemd denepdency

Let logiops start after multi-user.target is reached, and change it as a dependency to `graphical.target`. Fix #204