PixlOne/logiops #207: Suggestion: make a certain button (dpi button for example) act as a modifier

zany130, 2021-03-01:
So I was thinking it would be cool if holding the dpi button the back and forward buttons would do something different than when I just press them alone. I don't if this is already possible so sorry if I missed something.

c1p0, 2021-04-06:
This is something that I am interested in as well. I've gone through the documentation and examples and there doesn't seem to be any information about this, but maybe I'm missing something. Not sure if this can be done as a modifier or if it can be done when the driver detects the two buttons being pressed together, no matter the order. There might be an easy way to do this from the config file though, but I've yet to find out how.

Fish-Who, 2021-11-29:
This seems to be a function anywhere3 has on windows, when pressing forward, the scroll wheel act as thrumb wheel, wound be helpful if there would be modifier function or add such a action to gesture