PixlOne/logiops #211: Local configuration file and systemd

epassaro, 2021-03-03:
Hi!

`README.md` says using the `-c` flag we can specify a custom path for the configuration file. But how to use this in combination with `systemd`?

Thanks.

emilcarr, 2021-03-04:
From [man systemd.service](https://www.freedesktop.org/software/systemd/man/systemd.service.html)

```
ExecStart=
           Commands with their arguments that are executed when this service is started. The value is split into zero or more command lines according to the rules described below (see section "Command Lines"
           below)
```

Simply edit the systemd unit file in /lib/systemd/system/logid.service and append any arguments onto the end of the `ExecStart=` option. However, I would question why you wish to use an alternative config file from `/etc/logid.cfg` for a system-wide service as opposed to following UNIX convention.

emilcarr, 2021-03-04:
Suggesting this issue is closed

closed 2021-03-04 by epassaro