PixlOne/logiops #221: Configuring Thumbwheel actions

rad4day, 2021-03-26:
Hi,

I've already read in the wiki and a few issues related to the thumbwheel. I'm still not able to figure out how to configure the thumbwheel touch/tap/proxy actions. Currently I'm getting the following output from logid:
```
$ sudo logid -v DEBUG
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw12:255
[DEBUG] /dev/hidraw12:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[WARN] Line 117: Action type is missing, ignoring.
[WARN] Line 117: Invalid action
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```

My current config is quite long, so I'll attach it as a gist. I think my syntax is correct, but it looks like the action is dismissed as invalid before the thumb wheel is detected in the log.

[My Config](https://gist.github.com/rad4day/c654cc65f9c4fd372eb1cef1c52181b8)

I would really enjoy some help with debugging this problem.

Greetings
Tobias

rad4day, 2021-03-26:
Okay I have no idea what went wrong here. I locally rebuilt the program, ran it once and used the package I installed from AUR again... And it works? Did I just encounter some weird race condition? My config is unchanged to the diff above.

Just to make that very clear: I did not change my installed executable. So I'm still running the build that just failed loading the same config. And now it works...

ReillyBrogan, 2021-03-26:
@rad4day I just gave this a try and had no problem getting it working with the following:
```
        tap:
        {
            type: "Keypress"
            keys: ["KEY_A"]
        }
```

It does seem to take a few seconds after the logid service starts/restarts before the binding works.

rad4day, 2021-03-26:
Okay figured it out. It was actually me being unaware how vim and sudoedit work with each other. Apparently the original file is only overwritten when the editor is closed down.

So yeah. Sorry for the fuss.

closed 2021-03-26 by rad4day