PixlOne/logiops #222: Logiops Setup

mobiusstripe, 2021-03-29:
Hey, guys. There really isn't a complete setup guide here on the wiki. I install logiops through AUR, create the config file, and then what? And if there is are terminal commands to start logiops, can I make it so it stays on between restarts?

Thank you in advance,
mobius

ReillyBrogan, 2021-03-30:
@mobiusstripe Logiops builds a systemd unit file, so you can start it like so:
`sudo systemctl start logid`

Enable it on every boot with:
`sudo systemctl enable logid`

And restart it (after making changes to the config file):
`sudo systemctl restart logid`

mobiusstripe, 2021-04-01:
@ReillyBrogan It's still not working, but appreciate the help. This is my config file.

devices = (
{
    name = "Wireless Mouse MX Master 2S";
    smartshift =
    {
        on = true;
        threshold = 10; # 7 is ideal for work
    };
    hiresscroll =
    {
        hires = true;
        invert = false;
        target = true;
        up = {
            mode = "Axis";
            axis = "REL_WHEEL_HI_RES";
            axis_multiplier = 0.65;
        },
        down = {
            mode = "Axis";
            axis = "REL_WHEEL_HI_RES";
            axis_multiplier = -0.65;
        },
    };
    dpi = 1800;# <- you may change this number. 4000 is the maximum.

    buttons = (
        {
            cid = 0xc3;
            action =
            {
                type = "Gestures";
                gestures = (
                    {
                        direction = "Up";
                        mode = "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys = ["KEY_LEFTCTRL", "KEY_LEFTALT",  "KEY_L"];
                        };
                    },
                    {
                        direction = "Down";
                        mode = "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys = ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN"];
                        };
                    },
                    {
                        direction = "Left";
                        mode = "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys = ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_LEFT"];
                        };
                    },
                    {
                        direction = "Right";
                        mode = "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys = ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_RIGHT"];
                        }
                    },

                    {
                        direction = "None"
                        mode = "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys = ["KEY_WWW"];
                        }
                    }
                );
            };
        },
        {
            cid = 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            cid = 0x53;
            action = 
            {
                type = "Keypress";
                # keys = [KEYZOOMOUT];
                keys = ["KEY_LEFTCTRL", "KEY_MINUS"];
            };
        },
        {
            cid = 0x56;
            action = 
            {
                type = "Keypress";
                # keys = [KEY_ZOOMIN];
                keys = ["KEY_LEFTCTRL", "KEY_EQUAL"];
            };
        }
    );
}
);

ReillyBrogan, 2021-04-01:
Then see what the logs say:
`sudo journalctl -u logid.service -b --no-pager`

mobiusstripe, 2021-04-01:
@ReillyBrogan I'll try that out.

mobiusstripe, 2021-04-02:
@ReillyBrogan Turns out I just needed to setup a logid.service file. Thank you for your help! Much appreciated!

I'll paste my code here for anyone that needs to set one up. You can set up a config in ANY directory so, you don't have to mess with root permissions and the like.

[Unit]
Description=Logitech Configuration Daemon

[Service]
Type=simple
ExecStart=/usr/bin/logid -c *Your directory (/blah/blah)*
User=root
#ExecReload=/bin/kill -HUP $MAINPID

[Install]
WantedBy=multi-user.target

closed 2021-04-02 by mobiusstripe

ReillyBrogan, 2021-04-02:
@mobiusstripe How did you install logiops in the first place? Package from your distribution or by compiling it? And if you compiled it did you make sure to run `sudo make install` after you did that (as that step would have created a unit file for you)?

mobiusstripe, 2021-04-04:
@ReillyBrogan The former, a package from my distribution. I'm running Manjaro XFCE (for the first time, really) and they have the equivalent of an app store that can install AUR packages if enabled. I'm guessing it didn't make a logid.service fill when building logiops.