PixlOne/logiops #223: Not working right

thejoker3000, 2021-03-30:
I followed the guide lines in README.md and did the install on the dependencies needed. All good. Installation (make .... ) all works fine.
When running the final by doing `./logid` I get the following and I'm not quite sure why it doesn't work. I added the .cfg file in /etc/ but the buttons or volume wheel don't work:

```
~/logiops on master ?1 > sudo bash install.sh 
Installing dependencies now...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.16.3-1ubuntu1).
libconfig++-dev is already the newest version (1.5-0.4build1).
libevdev-dev is already the newest version (1.9.0+dfsg-1ubuntu0.1).
libudev-dev is already the newest version (245.4-4ubuntu3.5pop0~1615219913~20.04~d10e396).
The following packages were automatically installed and are no longer required:
  gir1.2-keybinder-3.0 libkeybinder-3.0-0 python3-configobj python3-psutil
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Building the tool now...
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
LogiOps Version Number: v0.2.2-52-g0a26579
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libevdev'
--   Found libevdev, version 1.9.0
-- Checking for module 'systemd'
--   Found systemd, version 245
-- Checking for module 'libconfig'
--   Found libconfig, version 1.5
-- Checking for module 'libudev'
--   Found libudev, version 245
-- systemd units will be installed at /lib/systemd/system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/thejoker/logiops/build
Scanning dependencies of target logid
[  1%] Building CXX object src/logid/CMakeFiles/logid.dir/logid.cpp.o
[  3%] Building CXX object src/logid/CMakeFiles/logid.dir/util/log.cpp.o
[  5%] Building CXX object src/logid/CMakeFiles/logid.dir/InputDevice.cpp.o
[  6%] Building CXX object src/logid/CMakeFiles/logid.dir/DeviceManager.cpp.o
[  8%] Building CXX object src/logid/CMakeFiles/logid.dir/Device.cpp.o
[ 10%] Building CXX object src/logid/CMakeFiles/logid.dir/Receiver.cpp.o
[ 11%] Building CXX object src/logid/CMakeFiles/logid.dir/Configuration.cpp.o
[ 13%] Building CXX object src/logid/CMakeFiles/logid.dir/features/DPI.cpp.o
[ 15%] Building CXX object src/logid/CMakeFiles/logid.dir/features/SmartShift.cpp.o
[ 16%] Building CXX object src/logid/CMakeFiles/logid.dir/features/HiresScroll.cpp.o
[ 18%] Building CXX object src/logid/CMakeFiles/logid.dir/features/RemapButton.cpp.o
[ 20%] Building CXX object src/logid/CMakeFiles/logid.dir/features/DeviceStatus.cpp.o
[ 21%] Building CXX object src/logid/CMakeFiles/logid.dir/features/ThumbWheel.cpp.o
[ 23%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/Action.cpp.o
[ 25%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/NullAction.cpp.o
[ 26%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/KeypressAction.cpp.o
[ 28%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/ToggleHiresScroll.cpp.o
[ 30%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/ToggleSmartShift.cpp.o
[ 31%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/CycleDPI.cpp.o
[ 33%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/ChangeDPI.cpp.o
[ 35%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/GestureAction.cpp.o
[ 36%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/ChangeHostAction.cpp.o
[ 38%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/Gesture.cpp.o
[ 40%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/ReleaseGesture.cpp.o
[ 41%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/ThresholdGesture.cpp.o
[ 43%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/IntervalGesture.cpp.o
[ 45%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/AxisGesture.cpp.o
[ 46%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/NullGesture.cpp.o
[ 48%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/Error.cpp.o
[ 50%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/raw/DeviceMonitor.cpp.o
[ 51%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/raw/RawDevice.cpp.o
[ 53%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/dj/Receiver.cpp.o
[ 55%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/dj/ReceiverMonitor.cpp.o
[ 56%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/dj/Error.cpp.o
[ 58%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp/Device.cpp.o
[ 60%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp/Report.cpp.o
[ 61%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp10/Error.cpp.o
[ 63%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp10/Device.cpp.o
[ 65%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/Device.cpp.o
[ 66%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/Error.cpp.o
[ 68%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/Feature.cpp.o
[ 70%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/EssentialFeature.cpp.o
[ 71%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Root.cpp.o
[ 73%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/FeatureSet.cpp.o
[ 75%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/DeviceName.cpp.o
[ 76%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Reset.cpp.o
[ 78%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/AdjustableDPI.cpp.o
[ 80%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/SmartShift.cpp.o
[ 81%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ReprogControls.cpp.o
[ 83%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/HiresScroll.cpp.o
[ 85%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ChangeHost.cpp.o
[ 86%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/WirelessDeviceStatus.cpp.o
[ 88%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ThumbWheel.cpp.o
[ 90%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/dj/Report.cpp.o
[ 91%] Building CXX object src/logid/CMakeFiles/logid.dir/util/workqueue.cpp.o
[ 93%] Building CXX object src/logid/CMakeFiles/logid.dir/util/worker_thread.cpp.o
[ 95%] Building CXX object src/logid/CMakeFiles/logid.dir/util/task.cpp.o
[ 96%] Building CXX object src/logid/CMakeFiles/logid.dir/util/thread.cpp.o
[ 98%] Building CXX object src/logid/CMakeFiles/logid.dir/util/ExceptionHandler.cpp.o
[100%] Linking CXX executable ../../logid
[100%] Built target logid
Creating systemctl now...
Failed to enable unit: Unit file logid.service does not exist.
Build is done. You can test it by running 'sudo ./logid'

~/logiops on master ?1 > sudo ./build/logid
[INFO] Detected receiver at /dev/hidraw1
[INFO] Device found: Wireless Mouse MX Master on /dev/hidraw1:1
[WARN] Error adding device /dev/hidraw2: std::exception
^C
~/logiops on master >
```
The script **install.sh** is something I made. I like automating things and made it simpler. I did a PR to add it and you can check it also.

mikeanth-dev, 2021-04-12:
Have you tried running it with `sudo`? As in, after running `make`, you run `sudo ./logiops` instead of `./logiops`. That worked for me

thejoker3000, 2021-04-12:
Yes, I have