PixlOne/logiops #226: Stops working after upgrade from Fedora 33 to 34 (Gnome 40)

MrTomRod, 2021-03-31:
I upgraded from Fedora 33 to 34 (`5.11.10-300.fc34.x86_64`) and `logiops` stopped working properly. I have a MX Master 2.

The service appears to work normally:

```
[root]# systemctl status logid
● logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; enabled; vendor preset: disabled)
     Active: active (running) since Wed 2021-03-31 10:02:00 CEST; 5min ago
   Main PID: 12581 (logid)
      Tasks: 7 (limit: 18691)
     Memory: 744.0K
        CPU: 1.248s
     CGroup: /system.slice/logid.service
             └─12581 /usr/bin/logid

Mar 31 10:02:00 user-fedora systemd[1]: Started Logitech Configuration Daemon.
```

The mouse itself works normally, but all the buttons/gestures I programmed have no effect.

When I run `logid` in verbose mode, I get the following output. No additional lines appear when I press buttons/perform gestures.

```
[root]# systemctl stop logid
[root]# logid -v
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[INFO] Device found: Wireless Mouse MX Master on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
```

My `/etc/logid.cfg`:

```
devices: (
{
    name: "Wireless Mouse MX Master";
    smartshift:
    {
        on: true;
        threshold: 30;
    };

    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                {
                    direction: "Left";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_LEFT"];
                    }
                },
                {
                    direction: "Right";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_RIGHT"];
                    }
                },
                {
                    direction: "Up";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_UP"];
                    }
                },
                {
                    direction: "Down";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN"];
                    }
                },
                {
                    direction: "None";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTMETA"];
                    }
                }
                );
            };
        }
    );
}
);
```

ReillyBrogan, 2021-04-01:
Does it work if you wait a minute or so after boot and restart it with `sudo systemctl restart logid`? If so your issue might be the same as #204.

go-ten, 2021-04-01:
Thank you for the answer.

The service is started at boot time (with sudo systemctl enable logid). A restart of the service (with sudo systemctl restart logid) does not change anything.

Everything was perfect under Fedora 33 and gnome 3.38.

Do you need more info ?

ReillyBrogan, 2021-04-01:
@go-ten No, that seems to be sufficient confirmation that it's not the same as the other issue as restarting it temporarily fixes that one. 

I'm not an actual logiops maintainer so I can't say what additional info would be helpful

MrTomRod, 2021-04-05:
Restarting does not help on my system either. Issue persists with `logiops-0.2.2-4.fc34.x86_64`.

jixbo, 2021-04-06:
Hello, I have the same issue after upgrading to Ubuntu 21.04 (beta).
The mouse settings, like dpi or scrolling work fine, but mapping the buttons to input event codes, even back and forward, is broken.
Previous version of Ubuntu was using kernel 5.8, and now it's 5.11, so something might have changed there.
I don't think it's related to gnome, as I have tried i3 with same issue. I have also tried gnome with xorg and wayland, but nothing worked.
Thank you.

jixbo, 2021-04-06:
I was having the same issue after upgrading to ubuntu 21.04 with kernel 5.11, mapping of the buttons had stopped working but the rest of the configuration (dpi, scrolling...) worked fine.

Updating logiops to the latest version, v0.2.3, fixed the issue for me.

MrTomRod, 2021-04-14:
Version 0.2.3 was [just released](https://bodhi.fedoraproject.org/updates/?search=logiops) in the Fedora 34 updates-testing repo.

It fixed the bug! (Simply install the update and run `sudo systemctl restart logid`.)

closed 2021-04-14 by MrTomRod