PixlOne/logiops #227: MX Anywhere 3 presents an extremely fast scroll speed on Arch Linux

VictorQueiroz, 2021-04-01:
When I enable high-resolution scroll, a small movement on mouse wheel results in a big scroll, which I believe to be unexpected. Is there anyway to fix this? I'm currently using Linux LTS kernel version (5.10.27-1). I also tried with the 5.11.11.arch1-1 version, but I get the same behavior.

Also, is it possible to change the scroll speed? I've tried changing settings on my MacBook Pro and switching back to Linux, but it didn't work. I could not find scroll speed property on the wiki.

Thank you very much for this library.

emvaized, 2021-04-05:
I'm facing the same issue with MX Master 3 when enabling hires scroll in cfg file - everything scrolls extremely fast, and rotating the wheel only for a few millimeters causes page to be scrolled by 40 lines (checked in Kate now).

KDE Neon 5.21
KDE Plasma 5.21.3
Linux 5.4.0-70 generic

emvaized, 2021-04-05:
@VictorQueiroz 
Seems that I found somewhat workaround - try to set config like this:
```
  hiresscroll: {
       hires: true;
       invert: false;
       target: true;
       up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 3;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -3;
        },
    };
```
I believe that `multiplier` values allow to adjust scroll speed

StefanLobbenmeier, 2021-05-10:
Interesting - I checked this issue to fix my slow scroll speed :stuck_out_tongue: the solution above also worked for me. I previously had the hiresscroll set to 

```
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
```

which resulted in even the "fast" spinning mode only scrolling the page very slowly.

Sewdohe, 2022-10-29:
> @VictorQueiroz Seems that I found somewhat workaround - try to set config like this:
> 
> ```
>   hiresscroll: {
>        hires: true;
>        invert: false;
>        target: true;
>        up: {
>             mode: "Axis";
>             axis: "REL_WHEEL_HI_RES";
>             axis_multiplier: 1;
>         },
>         down: {
>             mode: "Axis";
>             axis: "REL_WHEEL_HI_RES";
>             axis_multiplier: -1;
>         },
>     };
> ```
> 
> I believe that `multiplier` values allow to adjust scroll speed. Setting them to `1` and `-1` resolve in quite regular scroll speed

This fixed my insane scroll speed on Endeavour X11 as well! 

I also found that using Wayland KDE presents you with an option for scroll speed. You can lower the scroll speed for the mouse itself, then logiops virtual mouse input as well for another fix on Wayland

CharlesChiuGit, 2023-05-19:
> @VictorQueiroz Seems that I found somewhat workaround - try to set config like this:
>
> ```
>   hiresscroll: {
>        hires: true;
>        invert: false;
>        target: true;
>        up: { [...]

This doesn't seem working in v0.3.1

zrxdaly, 2023-07-21:
> @VictorQueiroz Seems that I found somewhat workaround - try to set config like this:
>
> ```
>   hiresscroll: {
>        hires: true;
>        invert: false;
>        target: true;
>        up: { [...]

Using this setting in V0.3.3 causes these two error at multiplier lines:
[WARN] Error at line 17: SettingTypeException
[WARN] Error at line 22: SettingTypeException

polair, 2023-09-13:
Hey all. I'm not sure exactly what's at the root of it but I was able to solve this for myself with a udev hwdb entry that sets "MOUSE_WHEEL_CLICK_COUNT". For some reason the default is totally insane. So, I put the following into the file:

/etc/udev/hwdb.d/70-logitech-mouse.hwdb

```
mouse:*:name:*Logitech*:
MOUSE_WHEEL_CLICK_ANGLE=1
MOUSE_WHEEL_CLICK_COUNT=1

*:name:*Master*
MOUSE_WHEEL_CLICK_ANGLE=1
MOUSE_WHEEL_CLICK_COUNT=1

evdev:name:*Master*:*
evdev:input:*Master*
evdev:bluetooth:*Master*
*bluetooth*:*Master*
input:*Master*:
input:name:*Master*:
MOUSE_WHEEL_CLICK_ANGLE=1
MOUSE_WHEEL_CLICK_COUNT=1

evdev:name:*Logitech*:*:
evdev:input:*Logitech*:
evdev:bluetooth:*Logitech*:
input:*Logitech*:
input:name:*Logitech*:
*bluetooth*:*Logitech*
MOUSE_WHEEL_CLICK_ANGLE=1
MOUSE_WHEEL_CLICK_COUNT=1
```

flanaman, 2023-10-20:
> @zrxdaly 
> 
> Using this setting in V0.3.3 causes these two error at multiplier lines: [WARN] Error at line 17: SettingTypeException [WARN] Error at line 22: SettingTypeException

In another issue thread I learned that you need to use floats in stead of ints and this error will go away https://github.com/PixlOne/logiops/issues/116#issuecomment-1583041962

---

> > @VictorQueiroz Seems that I found somewhat workaround - try to set config like this:
> > ```
> >   hiresscroll: {
> >        hires: true;
> >        invert: false;
> >        target: true;
> >        up: {
> >             mode: "Axis";
> >             axis: "REL_WHEEL_HI_RES";
> >             axis_multiplier: 1;
> >         },
> >         down: {
> >             mode: "Axis";
> >             axis: "REL_WHEEL_HI_RES";
> >             axis_multiplier: -1;
> >         },
> >     };
> > ```
> > 
> > 
> > I believe that `multiplier` values allow to adjust scroll speed. Setting them to `1` and `-1` resolve in quite regular scroll speed
> 
> This doesn't seem working in v0.3.1 and v0.3.2

@CharlesChiuGit, also from https://github.com/PixlOne/logiops/issues/116#issuecomment-916879599 I tried using `hires: false`, `target: true`, and `axis: "REL_WHEEL"` and found that it works on v0.3.3

pank-su, 2024-01-21:
> Using this setting in V0.3.3 causes these two error at multiplier lines: [WARN] Error at line 17: SettingTypeException [WARN] Error at line 22: SettingTypeException

Try float numbers in multiplier