PixlOne/logiops #229: Ratchet mode only for MX Master 3

seunjeon, 2021-04-07:
Hi, I would like to use "Ratchet mode" only for my MX Master 3 because of the scrolling issue. Can anyone help me to setup a configuration file please?

I tried to disable SmartShift but only Freespin mode was available. I set the maximum value for the SmartShift threshold but no luck.

My setting in the Windows is as follows:
[image]

Thanks!

KoduckShih, 2021-04-22:
Hi ~

You mean you don't want to use Free-spin mode?

**TRY THIS, if I understand you correctly.**

- Modify the `smartshift` optional object: 

  ~~~json
  smartshift:
  {
      //enable smartshift or free-spin, true for smartshift, false for free-spin
      on: true;
      
      //adjust this value til you get used to it.
      threshold: 10;
  };
  ~~~

seunjeon, 2021-04-22:
Hi KoduckShih, Thanks for your reply.

I tried to modify smartshift option as you described but it only allows free-spin or smartshift mode only.

For my understanding, to use the ratchet mode only, I need to setup **"Fixed scroll wheel mode" as Ratchet** but there are no such option in the configuration file. Increasing the smartshift threshold to 255 does not help either.

KoduckShih, 2021-04-23:
Er...i am a little bit confusing about what you mean.
For my understanding, you cannot fix a mode by `logiops`.
But you can switch between them, either SmartShift or Freespin.
***
In my configuration, I set the default mode to 'Freespin', and use the button behind the wheel to switch it to SmartShift or switch back.

seunjeon, 2021-04-23:
Hi, as you know, the ### SmartShift function is to automatically switch between ### "Ratchet" and ### "Freespin" mode based on wheel rotation speed.

But I would like to keep the Ratchet mode regardless of the wheel speed because of scrolling issue.

I could set this up in the Windows machine as I shown in the picture above (SmartShift: Disabled, Fixed scroll wheel mode: Ratchet).

But I cannot do this with logiops in Linux machine.

seunjeon, 2021-04-23:
I managed to setup by switching to [Solaar](https://github.com/pwr-Solaar/Solaar).

KoduckShih, 2021-04-23:
Aah... I see what you mean.
I tried to set  the threshold value to 255, then restart the logid service.
Now in Ratchet mode it doesn't switch to Freespin mode no matter how fast I turn the wheel.
Emm... I didn't even know this was possible in the past.

dscain, 2024-08-30:
@KoduckShih can you share the configuration you have? Setting 255 for me will not do the work unfortunatley.

dscain, 2024-08-30:
@seunjeon I am able to enable via Solaar, but the scrolling jitters. Do you have the same experience?

seunjeon, 2024-08-31:
@dscain hi, I've been using Solaar on Ubuntu 22.04, and so far, I haven't run into any jitter or other issues. I've used the MX Master 3 and 3S, as well as the Anywhere 3 and 3S, and everything's been working smoothly for me. I'm really curious about why that happens too.