PixlOne/logiops #233: MX Master 3 example file

druck13, 2021-04-15:
On the MX Master 3 this gives the following:-
- Horizontal scroll wheel zoom out (Ctrl Minus) and in (Ctrl Shift Equals)
- Gesture up cycles through 400, 1000, 1600 dpi
- Guesture down toggles hi res scroll
- Guesture left and right changes tab (Ctrl PageUp and Ctrol Page down)

Please feel free to add this to the project this if you want.

`devices: (
{
    name: "Wireless Mouse MX Master 3";
    dpi: 1000;
    smartshift:
    {
        on: true;
        threshold: 20;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    thumbwheel:
    {
        divert: true;
        invert: false;
        left:
        {
            threshold: 1;
            interval: 1;
            direction: "Left";
            mode: "OnInterval";
            action = { type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_MINUS"]; };
        };
        right:
        {
            threshold: 1;
            interval: 1;
            direction: "Right";
            mode: "OnInterval";
            action = { type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_LEFTSHIFT", "KEY_EQUAL"]; };
        };
    };

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    { direction: "Up";    mode: "OnRelease"; action = { type: "CycleDPI"; dpis: [400, 1000, 1600]; }; },
                    { direction: "Down";  mode: "OnRelease"; action = { type: "ToggleHiresScroll"; }; },
                    { direction: "Left";  mode: "OnRelease"; action = { type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"]; }; },
                    { direction: "Right"; mode: "OnRelease"; action = { type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"]; }; },
                    { direction: "None"   mode: "NoPress" }
                );
            };
        },
        {
            cid: 0xc4;
            action = { type = "ToggleSmartshift"; };
        }
    );
}
);
`

pietrygamat, 2021-04-22:
About using horizontal wheel for zooming - the proposed suggestion has one rather annoying flaw: zoom only works for a window that currently has focus, and not the one that your cursor is currently hovering over. For my personal use slightly different approach - bind `KEY_LEFTCTRL` to a thumbwel touch sensor. 
```
    thumbwheel:
    {
        divert: true;
        invert: false;      
        
        touch: {
	  type: "Keypress";
	  keys: ["KEY_LEFTCTRL"]; 
	};
    };
```
Note, I do not use the vertical scroll for zooming - I scroll with vertical wheel, while keeping thumb on the horizontal one.

For completeness I could also map left and right gestures to the RES_WHEEL axis (so that horizontal wheel works in vertical axis, but modified with CTRL key) and use horizontal wheel for zooming, but then, the CTRL modifier would still apply to my main wheel anyway. Plus I find using the main wheel for zooming more comfortable - it is more precise if the freespin is off.

Currently searching for a way to map CTRL + Horizonal Wheel to something usefull, as right now my thumb wheel is useless for any other purposes.

druck-pexip, 2021-04-23:
My initial thoughts were to try to convert the horizontal scroll wheel to produce vertical scroll + CTRL key, to avoid the window having to have focus, but I couldn't work out how to do it with logiops. It might be possible with Xbindkeys, but that will need more investigation.

gfduszynski, 2021-08-16:
I've give it a shot and as far as I can tell the best fix for me is just commenting out (ThumbWheel.cpp:152)
```
            if(scroll_action) {
                //scroll_action->press(true);
                scroll_action->move(direction * event.rotation);
            }
```

``press(true)`` resets the accumulator for axis in ``IntervalGesture.cpp``. 
Reset should only occur at the end of the movement, not during as is now.
Work good for me :)