PixlOne/logiops #239: Drop settings by lost connections

meugr, 2021-05-03:
Hi there!
Debian 10, KDE
If I switch my MX Master 3 on/off OR switch Bluetooth on/off, logid doesn't work correctly.
My settings is dropped and I need to run forced `systemd restart logid`

`systemd status logid` MAYBE write me `[WARN] Error ignored on detached thread/task: _readReport read failed: Input/output error` (like a #180 )

ArtisticZhao, 2021-09-06:
Same on Ubuntu 18.04 and the logid didn't report any WARN or ERROR.

ArtisticZhao, 2021-09-07:
@meugr I had a temporary solution.

Add an udev rules in `/etc/udev/rules.d/90-my.rules`
```
# /etc/udev/rules.d/90-my.rules
ACTION=="add", SUBSYSTEM=="bluetooth", RUN{program}="/bin/systemctl  restart logid.service"
```
and call 
```
> sudo udevadm control --reload
```

This will restart logid service when a new bluetooth device added.
If your mouse use the usb, you can modify "bluetooth" to "usb".

meugr, 2021-09-07:
@ArtisticZhao Thanks, nice temporary hack. It's absolutely working for me

dumbasPL, 2022-02-18:
A bit better solution:

```
ACTION=="add", SUBSYSTEM=="input", ATTRS{id/vendor}=="046d", RUN{program}="/bin/systemctl restart logid.service"
```

this will trigger **only** when a Logitech device is connected. The previously suggested rule will trigger on **any** Bluetooth device (including non-Logitech ones)

if you want to be more granular you can use `ATTRS{id/product}=="b019"` (MX Master 2S in this example) or `ATTRS{phys}=="00:11:22:33:44:55"` to filter by a specific mac address. `ATTRS{name}=="MX Master 2S Mouse"` is also an option

fredsco, 2022-02-21:
This did not solve my issue...

diegodario88, 2023-09-17:
> A bit better solution:
>
> ```
> ACTION=="add", SUBSYSTEM=="input", ATTRS{id/vendor}=="046d", RUN{program}="/bin/systemctl restart logid.service"
> ```
>
> this will trigger **only** when a Logitech device is [...]

That solve the issue for me.

kylechui, 2024-06-05:
I just updated my Linux kernel and I believe it changed how `udev` registers the MX Master connection event; [this commit](https://github.com/kylechui/dotfiles/commit/94dd5a974c0b073e20a4b19762fc03e5eb3df37e) is how I was able to resolve the issue.