PixlOne/logiops #247: Scoll works strange

ExposedCat, 2021-05-16:
Here is my config:
```
smartshift:
    {
        on: true;
        threshold: 200;
        default_threshold: 200;
    };
```
When I am trying to scroll UP many times, it starts scrolling too slow instead of switching to FreeSpin like when i scoll down many times. How can i fix it? It makes wheel unusable.

dokumanx, 2021-06-30:
Same issue here. I started to think returning my mx master 3. It has great futures but it cannot even perform basic scroll action. Beside of all the feature it should precisely scroll.

ExposedCat, 2021-06-30:
@dokumanx i fixed it. Check out my config at my sightly-linux repo

ExposedCat, 2022-06-24:
@PeterUstinox not sure if I fixed it or just used to it. With this config, for so long time I have no any issue, mouse just works perfectly including auto-switching to FreeSpeen mode. Maybe with some time my hands "understood" the distance and strength to scroll well.  
Anyway, the part of my config responsible for this is at [5-15 lines](https://github.com/ExposedCat/dotfiles/blob/58056a8a114b9c0454a7f419be0a526a9a7eac32/logiops/logid.cfg#L5):
```
smartshift:
{
    on: true;
    threshold: 255;
};
hiresscroll:
{
    hires: true;
    invert: false;
    target: false;
}
```

ExposedCat, 2022-06-24:
@PeterUstinox not sure if I fixed it or just used to it. With this config, for so long time I have no any issue, mouse just works perfectly including auto-switching to FreeSpeen mode. Maybe with some time my hands "understood" the distance and strength to scroll well.  
Anyway, the part of my config responsible for this is at [5-15 lines](https://github.com/ExposedCat/dotfiles/blob/58056a8a114b9c0454a7f419be0a526a9a7eac32/logiops/logid.cfg#L5):
```
smartshift:
{
    on: true;
    threshold: 255;
};
hiresscroll:
{
    hires: true;
    invert: false;
    target: false;
}
```

Feel free to re-open the issue if you can reproduce the problem even **after a ~month of using your mouse**.

closed 2022-06-24 by ExposedCat

PeterUstinox, 2022-06-26:
thanks, that was my first day with the mouse.
just sad, that it does not work intuitively then :(

but thanks! i use those settings an will see!