PixlOne/logiops #248: Speed up scroll with "target: false"

ExposedCat, 2021-05-16:
`hirescroll { target: true; }` makes scrolling strange, but `axis_multiplier` does not work without it. So is there any way to speed up scroll without target: true?

emvaized, 2021-06-23:
@ExposedCat 
Yes, you could do it like this:
```
  hiresscroll: {
       ...
        up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 2;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -2;
        },
    };
```
By adjusting `axis_multiplier,` you can change the scroll speed

ExposedCat, 2021-06-23:
@emvaized then i need to enable `target: true`. Otherwise multiplier will not work. The only way i see for now is IMWheel