PixlOne/logiops #249: Could not create input device: No such device

quantum-booty, 2021-05-17:
I am following the archlinux guide https://wiki.archlinux.org/title/Logitech_MX_Master for my mx master 3.
I first created this logid.cfg
```
devices: ({
  name: "Wireless Mouse MX Master 3";

  // A lower threshold number makes the wheel switch to free-spin mode
  // quicker when scrolling fast.
  smartshift: { on: true; threshold: 20; };

  hiresscroll: { hires: false; invert: false; target: false; };

  // Higher numbers make the mouse more sensitive (cursor moves faster),
  // 4000 max for MX Master 3.
  dpi: 400;

  buttons: (

    // Make thumb button 10.
    { cid: 0xc3; action = { type: "Keypress"; keys: ["BTN_FORWARD"]; }; },

    // Make top button 11.
    { cid: 0xc4; action = { type: "Keypress"; keys: ["BTN_BACK"];    }; }

  );
});
```
Then, when trying to obtain the device name using ```logid -v```, it gives me this error:
```[ERROR] Could not create input device: No such device```.

The mouse is working fine, so I don't know why its giving me this errro.

Vallefor, 2021-05-18:
Have the same error on manjaro.

quantum-booty, 2021-05-18:
hmm, it somehow automatically fixed itself, idk what I did. I rebooted my system (arcolinux), but idk if this is the reason. I'll leave this open for Vallefor.

Vallefor, 2021-05-18:
Also started to work after reboot. I 99% sure, that the problem appear after upgrade system with kernel and decline reboot.
@quantum-booty you can close the issue :)

closed 2021-05-18 by quantum-booty