PixlOne/logiops #251: config not being applied.

zany130, 2021-05-19:
version: aur/logiops 0.2.3-1 (+6 0.79)
logs:
```
zany130@kiseki /@pkgcache 🦇» logid -v                                                           14:09
[DEBUG] Unsupported device /dev/hidraw8 ignored
[DEBUG] Unsupported device /dev/hidraw11 ignored
[DEBUG] Unsupported device /dev/hidraw10 ignored
[DEBUG] Unsupported device /dev/hidraw13 ignored
[DEBUG] Unsupported device /dev/hidraw9 ignored
[DEBUG] Unsupported device /dev/hidraw12 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
[WARN] Error adding device /dev/hidraw1: std::exception
[INFO] Detected receiver at /dev/hidraw0
[INFO] Device Wireless Mouse MX Master 3 not configured, using default config.
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw0:1
[DEBUG] /dev/hidraw0:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```
Config:
```
cat /etc/logid.cfg                                                 14:15
devices: (
{
    name: "Logitech Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			   Keys:  ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_K"];

                        };
                    },

                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			   Keys:  ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_H"];

                        };
                    },

                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_L"];

                        }
                    },

                    {
                        direction: "None"
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_ENTER"];

                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
    );
},
{
# Another device to configure
name: "Other Logitech USB Receiver";

}
);
```

zany130, 2021-05-22:
forgot to run it as sudo above but same error happens 

```
sudo logid -v                                                       9:31
[sudo] password for zany130: 
[DEBUG] Unsupported device /dev/hidraw8 ignored
[DEBUG] Unsupported device /dev/hidraw9 ignored
[DEBUG] Unsupported device /dev/hidraw10 ignored
[DEBUG] Unsupported device /dev/hidraw11 ignored
[DEBUG] Unsupported device /dev/hidraw12 ignored
[DEBUG] Unsupported device /dev/hidraw13 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
[WARN] Error adding device /dev/hidraw1: std::exception
[INFO] Detected receiver at /dev/hidraw0
[ERROR] Caught HID++ 2.0 error while trying to initialize /dev/hidraw0:1: Invalid argument
```

brozjak2, 2021-05-24:
I suppose you have copied the config from Arch wiki. It is probably outdated.

Try changing the device name to only `"Wireless Mouse MX Master 3"`, as it is display in the output of `logid -v`:

```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    ...
```

That worked for me.

zany130, 2021-05-24:
> I suppose you have copied the config from Arch wiki. It is probably outdated.
> 
> Try changing the device name to only `"Wireless Mouse MX Master 3"`, as it is display in the output of `logid -v`:
> 
> ```
> devices: (
> {
>     name: "Wireless Mouse MX Master 3";
>     ...
> ```
> 
> That worked for me.

tried that still getting errors: 
```
● logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; enabled; vendor preset: disabled)
     Active: active (running) since Mon 2021-05-24 13:35:58 EDT; 3min 3s ago
   Main PID: 14985 (logid)
      Tasks: 7 (limit: 19096)
     Memory: 784.0K
        CPU: 0
     CGroup: /system.slice/logid.service
             └─14985 /usr/bin/logid

May 24 13:35:58 kiseki systemd[1]: Started Logitech Configuration Daemon.
May 24 13:36:00 kiseki logid[14985]: [WARN] Error adding device /dev/hidraw2: std::exception
May 24 13:36:01 kiseki logid[14985]: [WARN] Line 27: keys is a required field, skipping.
May 24 13:36:01 kiseki logid[14985]: [WARN] Line 37: keys is a required field, skipping.
May 24 13:36:01 kiseki logid[14985]: [WARN] Line 48: keys is a required field, skipping.
```
config :

```

cat /etc/logid.cfg
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			   Keys:  ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_K"];

                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			   Keys:  ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_J"];

                        };
                    },

                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			   Keys:  ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_H"];

                        };
                    },

                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_L"];

                        }
                    },

                    {
                        direction: "None"
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_ENTER"];

                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
    );
},
{
# Another device to configure
name: "Other Logitech USB Receiver";

}
);
```

auipga, 2022-11-10:
`Keys` != `keys`. A single **K** kills my file too.