PixlOne/logiops #254: Failed to start logid.service: Unit logid.service is masked.

ibrag474, 2021-06-04:
I installed and tried to ran it logiops by running this commands:
```
sudo apt install cmake libevdev-dev libudev-dev libconfig++-dev git
git clone https://github.com/PixlOne/logiops.git
cd logiops
mkdir build
cd build
cmake ..
make
sudo make install
sudo systemctl start logid
```
Systemctl threw this error:
`Failed to start logid.service: Unit logid.service is masked.`

fgsalomon, 2022-08-24:
The same happened to me, I think, because I'd previously installed and uninstalled `logiops` through `apt`.

I've checked that the unit was pointing to `/dev/null` with `file`:

```
file /etc/systemd/system/logid.service 
/etc/systemd/system/logid.service: symbolic link to /dev/null
```
Then I've removed the symbolic link and executed `sudo systemctl enable --now logid` and it worked.

Hope it helps.

ExposedCat, 2022-11-23:
Can be fixed just by `systemctl unmask logid`