PixlOne/logiops #256: Key remapping stop working in Kernel 5.12

leodream, 2021-06-06:
The tool no longer remaps the key afterr I upgraded the kernel to **5.12.4-2**  and the tool to **v0.2.3** (checked out v0.2.3 and recompiled. invoke with "-V" shows "v0.2.2-52-g0a26579").  

Debug output
```
sudo logid -c /etc/logid.cfg -v                                                                                          
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw2 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
[INFO] Device Wireless Mouse MX Master 3 not configured, using default config.
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw8:255
[DEBUG] /dev/hidraw8:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```

Config file
```
devices: (
{
    name: "Wireless Mouse MX Master";
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: true;
    };
    dpi: 1600;
    buttons: (
        {
            cid: 0x53;
            action =
            {
                type: "Keypress";
                keys: ["KEY_PAGEUP"];
            };
        },
        {
            cid: 0x56;
            action =
            {
                type: "Keypress";
                keys: ["KEY_PAGEDOWN"];
            };
        }

    );
}
);                                                                                                                                                                                    
```

Tried with different cids and keys combinations, never works. It looks like the buttons got completely ignored. There is no issue on the dpi, hiresscroll etc. settings though.

Have anyone encounterred similar issues?

foursixnine, 2021-07-25:
@leodream you need to set the device name to `Wireless Mouse MX Master 3` for the config file to be loaded

leodream, 2021-07-27:
Sorry to bother, should pay more attention to the log.

closed 2021-07-27 by leodream