PixlOne/logiops #262: Howto remap the ToggleSmartShift button as middle button of a MX Anywhere 3

thhart, 2021-07-29:
Sorry to ask this frankly but the configuration is a bit complicated and unfortunately there are very few samples. Maybe it is a good idea to add some default sample ones into the wiki?!
I have a MX 3 Anywhere, cool mouse but unfortunately Logitech changed its default settings back, so the middle button is mapped to the wheel by default and the button at the wheel is toggling the SmartShift behavior, this is bad by design.
So I am asking directly how to revert this to old MX 2 style (middle button to the center button instead of the wheel), a sample config would be nice.

oscarpax, 2021-07-30:
I also have a MX Anywhere 3 and I would also appreciate a sample config of how to map the middle button to the center button and ratchet toggle to the wheel.

I can achieve this with libratbag, but settings reset when it goes to sleep or is turned off and I have to re-apply the settings.

thhart, 2021-08-06:
Ok, here we are, the following config is working for me. It is simply toggling the buttons back to older MX style which is much better.
Please note the following patch is necessary to fully support smartshift in the MX Anywhere 3: https://github.com/PixlOne/logiops/issues/174#issuecomment-884343407
Further the config file contains a customized scroll speed adjustment, change axis_multiplier for your own.

```
devices: (
{
    name: "MX Anywhere 3";
	smartshift:
    	{
        on: false;
        threshold: 30;
    	};
	hiresscroll: {
       hires: true;
       invert: false;
       target: true;
       up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 2;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -2;
        },
    };

    buttons: (
		{
            cid: 0x52;
            action =
            {
                type: "ToggleSmartshift";
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "Keypress";
                keys: ["BTN_MIDDLE"];
            };
        }
    );
}
);
```

oscarpax, 2021-08-07:
@thhart Thanks!

Shamus03, 2023-05-05:
I had this working on a previous computer and now I can't get it to work as of the latest version.  It seems like any attempt to remap 0xc4 does not work.  Here is my simple config file:
```
devices: (
{
    name: "MX Anywhere 3";
    smartshift:
    {
        on: true;
        threshold: 7;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 800;

    buttons: (
        {
            cid: 0xc4;
            action =
            {
                type = "Keypress";
                keys: ["BTN_MIDDLE"];
            };
        }
    );
}
);
```

I switched to the commit I had used on my old machine, dbe2b28a5307dfc25e7b4858b94bded9acc5ea5c, and rebuilt/reinstalled and it's working fine.  I can deal with that for now, but I'm wondering what core thing changed from then until now to make it stop working.

PixlOne, 2023-05-06:
Could you post output of `sudo logid -v DEBUG`?

Shamus03, 2023-05-08:
> Could you post output of `sudo logid -v DEBUG`?

I do not want to uninstall my (working) installation of logiops so I just rebuilt it and am running it from the build folder.  Confirmed the button remapping went back to not working:

```
build:main λ sudo ./logid -v DEBUG
[DEBUG] Ignoring virtual node on /dev/hidraw1
[INFO] Detected receiver at /dev/hidraw0
[INFO] Device found: MX Anywhere 3 on /dev/hidraw0:2
[DEBUG] /dev/hidraw0:2 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
```

If I switch back to the commit I mentioned and rebuild and rerun, I get almost the same output, but with one exception (and my mouse rebinding works):

```
build λ git status
HEAD detached at dbe2b28
nothing to commit, working tree clean
build λ sudo ./logid -v DEBUG
[INFO] Detected receiver at /dev/hidraw0
[INFO] Device found: MX Anywhere 3 on /dev/hidraw0:2
[DEBUG] /dev/hidraw0:2 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[WARN] Error adding device /dev/hidraw1: std::exception
```