PixlOne/logiops #265: Logid systemd service doesn't work on boot (Manjaro/Arch)

Maikuh, 2021-08-21:
On every boot, the service fails to start, below you can see the log:
```
Aug 18 17:23:49 mangy systemd[1]: Started Logitech Configuration Daemon.
Aug 18 17:23:51 mangy logid[523]: [WARN] Error adding device /dev/hidraw2: open failed: No such file or directory
Aug 18 17:23:52 mangy logid[523]: [WARN] Error adding device /dev/hidraw1: std::exception
Aug 18 17:23:52 mangy logid[523]: [WARN] Error adding device /dev/hidraw0: Invalid sub ID
Aug 18 17:23:54 mangy logid[523]: [WARN] Error adding device /dev/hidraw1: std::exception
Aug 18 17:23:54 mangy logid[523]: [WARN] Error adding device /dev/hidraw1: std::exception
Aug 18 13:36:11 mangy logid[523]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw0:1: Resource error
Aug 18 13:36:11 mangy logid[523]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw0:1: Resource error
Aug 18 13:38:14 mangy logid[523]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw0:1: Resource error
Aug 18 13:38:41 mangy logid[523]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw0:1: Resource error
Aug 18 13:39:02 mangy logid[523]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw0:1: Resource error
Aug 18 13:39:03 mangy logid[523]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw0:1: Resource error
Aug 18 13:40:38 mangy logid[523]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw0:1: Resource error
Aug 18 14:32:11 mangy systemd[1]: Stopping Logitech Configuration Daemon...
Aug 18 14:32:12 mangy systemd[1]: logid.service: Deactivated successfully.
```

This only happens on boot, since whenever I restart the service (using `sudo systemctl restart logid`) or run logid via terminal (like `logid -v`) it works fine until I reboot again and have to manually restart it.

I'm currently using Manjaro GNOME 21.1.0, but I was using Pop-OS 21.04 until recently and this didn't happen.

avasyub, 2021-11-23:
This issue seems to be fixed in master (https://github.com/PixlOne/logiops/pull/246), but is not yet released as a version.

Try using logiops-git from AUR or build it from source.

I've been using logiops-git for the past few days and logid starts up just fine on boot. No restarting needed.

chomosuke, 2022-12-02:
> This issue seems to be fixed in master (#246), but is not yet released as a version.
> 
> Try using logiops-git from AUR or build it from source.
> 
> I've been using logiops-git for the past few days and logid starts up just fine on boot. No restarting needed.

Hmmm for me restart is still needed every single time I boot. I'm using the latest master commit. I can provide more info if needed.

ocroquette, 2024-05-15:
I am still having this issue. I am on this commit:

```
commit 94f6dbab5390c1c7375836dd9314c0c2488e48a3 (tag: v0.3.3, upstream/main)
Merge: 31d1955 f2680bd
Date:   Wed Jul 12 17:39:21 2023 -0400

    Merge pull request #399 from PixlOne/cmake-fixes
    
    CMake fixes: NDEBUG in None build type and allow shared ipcgull library
```

Here is the log file:
```
-- Boot c8b7ce67b2d64f3692541bd144c0c826 --
May 15 07:06:42 host systemd[1]: Started Logitech Configuration Daemon.
May 15 07:53:52 host  systemd[1]: Stopping Logitech Configuration Daemon...
May 15 07:53:53 host  systemd[1]: logid.service: Deactivated successfully.
May 15 07:53:53 host  systemd[1]: Stopped Logitech Configuration Daemon.
May 15 07:53:53 host  systemd[1]: Started Logitech Configuration Daemon.
```

After the boot at 07:06 logid did not work. After a restart of the service at 07:53, it did. I had the problem already on various versions of Ubuntu. I have it too on Manjaro, right now.

ocroquette, 2024-05-26:
The problem became worst lately. logid stops working regularly, for instance after the system is idle for a while (not necessarily in standby, just unused). Restarting the service always fixes it, but still, it is a bit annoying. There is nothing useful in the log files.

Looking at the comments in https://github.com/PixlOne/logiops/pull/308, it looks like the project is not maintained anymore, unfortunately.

PixlOne, 2024-05-28:
Sorry, I haven't really had the time to contribute to this project lately. Unfortunately I haven't really looked at the recently posted issues or pull requests made. In the coming weeks, I'll try to review the current urgent issues that prevent the program from operating normally. If you could help point to some urgent ones, that would be great, thanks.

ocroquette, 2024-05-28:
No worries, it is nice to hear from you :) On a scale from 1 (showstopper) to 3 (minor), this issue would be for me a 2. It does not prevent to use it, but having to restart the service manually on a daily basis is annoying. Let use know if we can do anything else on our end. Thanks!

hbarcelos, 2024-06-12:
I second this.
It happens to me too: any time the devide is disconnected by either turning it off or it becoming idle, so the OS disconnects it or even after a suspension of the system, the configured service does not work properly, and I need to restart the it to get it back.

hbarcelos, 2024-06-12:
Not sure how useful this is, but running `journalctl -xe` gives me:
```
Jun 12 14:40:49 a70mob logid[282434]: [INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw3:255
Jun 12 14:40:50 a70mob kernel: input: LogiOps Virtual Input as /devices/virtual/input/input59
Jun 12 14:40:50 a70mob touchegg[1120]: Warning: Error opening device /dev/input/event21
Jun 12 14:40:50 a70mob systemd-logind[1117]: Failed to open /dev/input/event21: No such device
Jun 12 14:40:50 a70mob kernel: input: LogiOps Virtual Input as /devices/virtual/input/input60
Jun 12 14:40:50 a70mob systemd-logind[1117]: Watching system buttons on /dev/input/event21 (LogiOps Virtual Input)
Jun 12 14:40:51 a70mob kernel: input: LogiOps Virtual Input as /devices/virtual/input/input61
Jun 12 14:40:51 a70mob systemd-logind[1117]: Watching system buttons on /dev/input/event21 (LogiOps Virtual Input)
Jun 12 14:40:52 a70mob logid[282434]: [INFO] Device found: MX Keys Wireless Keyboard on /dev/hidraw2:255
Jun 12 14:40:54 a70mob kernel: input: LogiOps Virtual Input as /devices/virtual/input/input62
Jun 12 14:40:54 a70mob systemd-logind[1117]: Failed to open /dev/input/event21: No such device
Jun 12 14:40:54 a70mob touchegg[1120]: Warning: Error opening device /dev/input/event21
Jun 12 14:40:54 a70mob kernel: input: LogiOps Virtual Input as /devices/virtual/input/input63
Jun 12 14:40:54 a70mob systemd-logind[1117]: Watching system buttons on /dev/input/event21 (LogiOps Virtual Input)
```

ocroquette, 2025-01-12:
There were quite some long-lasting issues with the Mediatek bluetooth drivers recently, like https://bugzilla.kernel.org/show_bug.cgi?id=219514 for instance, which is fixed in 6.12.8. After the kernel update, I do not need to restart the logid service anymore (or the bluetooth service). So some of the reports about logid are probably due to general bluetooth issues.