PixlOne/logiops #268: Binding mouse gesture to a touchpad gesture

AMDG2, 2021-09-06:
Is it possible to bind a mouse gesture to a touchpad gesture?

My use case is to use the MX Master 2s thumb button to emulate three-fingers gestures to allow switching between workspaces and overview in Gnome.

If it is possible right now, it would be nice to be able to bind a button gesture a touchpad with a configuration like this one:

```
 buttons: (
  {
    cid = 0xc3;
    action = 
      {
        type: "Gestures";
        gestures: (
        {
          direction: "All";
          action: {
            type: "TouchpadGestures";
            fingers: 3;
            mode: "Move"; # Drag, Pinch or other "kind" or gestures
          }
        }
        )
```

S-trace, 2021-12-06:
> did you get this to work?

It seems @AMDG2 have not made this working, this is just a concept.

Sewdohe, 2022-11-20:
This isn't possible ATM but you can do the same thing by defining a keyboard shortcut for your window manager and then re-creating that shortcut in logiops for the gesture. It works really well! 

If we could get 1:1 mappings like with touchegg or wayland working though that would literally be amazing

PixlOne, 2023-05-11:
I'm interested in adding this, will investigate later. Thanks for bringing it to my attention.