PixlOne/logiops #269: Error on first connection after boot

yursan9, 2021-09-17:
I have MX Master 3, whenever I connect the device after boot it always output the following message in `systemctl status logid`:
```
Error adding device /dev/hidraw2: Invalid function ID
```

So I try to run logid outside of systemd service, and I see that logid report wrong device name on first connection after reboot:
```
[INFO] Device Wireless Mouse MWireless M not configured, using default config.
[INFO] Device found: Wireless Mouse MWireless M on /dev/hidraw2:255
[INFO] Device Wireless Mouse MWireless M not configured, using default config.
[INFO] Device found: Wireless Mouse MWireless M on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x00 |         |         |            | 
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x00 |         |         |            | 
[WARN] Error adding device /dev/hidraw2: Invalid function ID
[WARN] Error adding device /dev/hidraw2: Invalid function ID
```

To reproduce you need to run logid before connecting (moving) the mouse. If I exit logid, and rerun the program, it will report the correct device name.

I already tried using the latest commit from master, but the problem still persist.

araujorm, 2022-01-11:
To me this seems to be 100% reproducible at each reboot. Any more info needed to fix it? Or is there any feasible workaround?

araujorm, 2022-01-13:
I decided to compare the modules that are present before and after I turn on the mouse. I found out that (at least on kernel 5.15.12) hid_logitech_hidpp is loaded. I then ensured that hid_logitech_hidpp was always loaded *before* logid, and that seems to workaround the issue.

For now one can do that via a conf file in /etc/modules-load.d or (as I do at the moment) with a systemd override (`sudo systemctl edit logid.service`) such as:

```
[Service]
ExecStartPre=/usr/sbin/modprobe hid_logitech_hidpp || :
```

Can anyone with the issue try it?

Anyway, this is just a workaround, the issue should be tackled by logid. My guess is that logid starts trying to do certain stuff that depend on that module as soon as it detects de device, and sometimes that happens before the module has had time to fully load (or even while it's still loading). That's why it only happens at boot, and it doesn't always happen (to me it seems to happen mostly when using my laptop on battery, when it is slower, for example)

@PixlOne , could there be anything that could be probed before trying to act on the device, to ensure this does not happen?

yursan9, 2022-01-14:
I have my own workaround. I use udev rule to restart `logid` on udev BIND event.
But I can try your solution @araujorm

Chr1s70ph, 2022-03-23:
> For now one can do that via a conf file in /etc/modules-load.d or (as I do at the moment) with a systemd override (`sudo systemctl edit logid.service`) such as:
> 
> ```
> [Service]
> ExecStartPre=/usr/sbin/modprobe hid_logitech_hidpp || :
> ```
> 
> Can anyone with the issue try it?

Editing the override file with `sudo systemctl edit logid.service` only causes errors (at least for me). I have edited the service file from hand now. I have not yet tested it with a restart, but do people even still run in this issue, or is it just me?

lazarosgogos, 2023-07-31:
> 

This worked for me as well. Thanks a lot friend !! <3

lazarosgogos, 2023-10-07:
> I decided to compare the modules that are present before and after I turn on the mouse. I found out that (at least on kernel 5.15.12) hid_logitech_hidpp is loaded. I then ensured that [...]

This fix worked for me! Thanks!!

lazarosgogos, 2024-07-22:
> I decided to compare the modules that are present before and after I turn on the mouse. I found out that (at least on kernel 5.15.12) hid_logitech_hidpp is loaded. I then ensured that [...]

This fix does not work anymore for me. Anyone else having an issue?

Logiops works after reboot, but not after any disconnect-and-reconnect, or log-out and log-back-in, or any `systemctl restart logid.service`

Any input can prove useful :)

Btw, this used to work before updating bluetooth from bluez-5.69 to bluez-5.77, from Fedora 39 to 40