PixlOne/logiops #271: Hires scrolling behaviour seems to have changed after upgrading to linux 5.14

johnhamelink, 2021-09-21:
Hi there,

I just upgraded to linux 5.14, and now I'm noticing a different scrolling behaviour when scrolling through the tabs in Qutebrowser.

When I have the following settings:

```
devices: ({

  name: "Wireless Mouse MX Master 2S";

  // A lower threshold number makes the wheel switch to free-spin mode
  // quicker when scrolling fast.
  smartshift: { on: true; threshold: 20; };

  hiresscroll: { hires: true; invert: false; target: false; };

  // Higher numbers make the mouse more sensitive (cursor moves faster),
  // 4000 max for the MX Master 3.
  dpi: 1500;

  buttons: (
    // Make top button (0xc4) switch modes
    { cid: 0xc4; action = { type = "ToggleSmartShift"; }; },
  );

});
``` 

Scrolling down a webpage feels responsive and correct, however when I try to scroll across tab bars, the scrolling is unusably sensitive. Inverting the `hiresscroll` settings produces the opposite result:

```
devices: ({

  name: "Wireless Mouse MX Master 2S";

  // A lower threshold number makes the wheel switch to free-spin mode
  // quicker when scrolling fast.
  smartshift: { on: true; threshold: 20; };

  hiresscroll: { hires: false; invert: false; target: false; };

  // Higher numbers make the mouse more sensitive (cursor moves faster),
  // 4000 max for the MX Master 3.
  dpi: 1500;

  buttons: (
    // Make top button (0xc4) switch modes
    { cid: 0xc4; action = { type = "ToggleSmartShift"; }; },
  );

});
```

Scrolling across tabs now feels correct, but scrolling down web pages feels slow and unresponsive. This issue seems to be new for linux 5.14, as I've been using logid with 5.13.x for around a year with no problems (except for needing to restart it now and then).

Pistrie, 2021-09-22:
Same here. Hires now also picks up scroll movement before a 'click' of the scroll wheel has been made. It detects movement of the scroll button in between these clicks.

johnhamelink, 2021-09-24:
@Pistrie yes that's true, and I think this might be the problem - the increased sensitivity seems to now be registered as a scroll event, whereas the click was the scroll event before. That explains why scrolling a page feels smooth (lots of events = responsive and smooth), but tab bar scrolling is now wildly sensitive because it's state change is activated by a single scroll event.

bkleiner, 2021-09-24:
i seem to be having the exact same issue as @Pistrie the scroll wheel movement outside of the "clicks" is picked up in ratchet mode. however i am not completely convinced it is related to the kernel version - i downgraded to 5.10.x ("lts" kernel on arch linux) and i am still seeing the same behaviour

chic-luke, 2021-09-26:
Bump. I am seeing the same exact issue on my MX Master 2s, both on the latest and lts kernel on Arch. However, I have not switched to LogiOps yet, still using Solaar for the moment. What else could this behaviour be caused by, if it's neither on the kernel nor on LogiOps?

chic-luke, 2021-09-29:
I did some digging and found the cause of the issue to be an update to `libinput`. In my specific case, I downgraded the package `xf86-input-libinput` from version 1.2.0 to 1.1.0 on my Arch install and, after a reboot, the smart shift button and functionality once again works as intended, so scrolling is usable.

It appears this one is on `libinput`

johnhamelink, 2021-09-29:
@chic-luke fantastic, yes downgrading to 1.1.0 did fix this for me - I must've updated my kernel and this package at the same time.

For me, the correct command to get back down to 1.1.0 was:

```
sudo pacman -U /var/cache/pacman/pkg/xf86-input-libinput-1.1.0-1-x86_64.pkg.tar.zst
```

Then in `/etc/pacman.conf`, in order to make sure that an upgrade doesn't put us back to 1.2.0 again:

```
IgnorePkg   = xf86-input-libinput
```

closed 2021-09-29 by johnhamelink