PixlOne/logiops #275: Include note in the wiki CIDs list regarding the "middle" button that might actually be the gesture button

gyohza, 2021-11-03:
I own an MX Anywhere 2S and I spent ages trying to figure out what the CID for the gesture button is.

[Turns out it was actually the middle button's](https://github.com/PixlOne/logiops/pull/127#issuecomment-958670775), but it took me quite a while to think that it might be it and actually try it out.

Maybe I was slow in figuring this out, but can we please update the wiki [CIDs list](https://github.com/PixlOne/logiops/wiki/CIDs) so that nobody else gets stuck in this little gotcha?

PixlOne, 2021-11-03:
Sure, just added that.

closed 2021-11-03 by PixlOne

seanmamasde, 2024-02-05:
@gyohza Hello, may I ask you how do you figure out the CIDs of your mouse? I own an MX Master 3 and want to see which CIDs each buttons/functions corresponds to. 
What I have tried is to run `tcpdump -i bluetooth0 -w ble.pcap` to capture the packets from the BLE interface, which only connects to my mouse while I was recording. I have also set `bluetoothctl scan on` in another terminal window before running `tcpdump`. I moved around a bit and clicked a few left clicks. Then I open up the ble.pcap file with wireshark, but all the packets with ATT protocol are with an unknown handle (`0x0033`) ([screenshot](https://imgur.com/cci6oRx)). Do you know why is this happening? Appreciate the help.

gyohza, 2024-02-22:
Hey, @seanmamasde - to be quite honest, I hadn't found a quick way to list my device's CIDs. **It looks like there's an open request [here](https://github.com/PixlOne/logiops/issues/113#issuecomment-670962331)**, with a comment from the maintainer:

> I'll try to add a debug action and use that to help people discover the right CIDs.

Although can't be sure where at which stage we're at for this feature, since we haven't had any updates there since 2020.

As for what I did back then, I think I figured it out by elimination. I no longer have the script, nor do I have logiops working and installed on my machine right now, but if memory serves, what I did was include each and every CID listed in the [CID table](https://github.com/PixlOne/logiops/wiki/CIDs) in my [config](https://github.com/PixlOne/logiops/blob/main/logid.example.cfg), e.g.:

```libconfig
    buttons: (
        {
            cid: 0x50;
            action =
            {
                type: "None";
            };
        },
        {
            cid: 0x51;
            action =
            {
                type: "None";
            };
        },
        # and so on
    );
```

Then I executed `logid` from a script. `logid` throws a warning if a given CID is not mapped for the device:
> [WARN] Wireless Mobile Mouse MX Anywhere 2S: CID 0xc3 does not exist

By eliminating the CIDs for which warnings had been thrown, I was then left with a list of valid CIDs. I guess there's a chance that a CID might not be in the table at all, but they all seem to be hexes lying between `0x50` and `0xfe`, so you can just loop through the values in between to generate a config with all the buttons. To save you some time:

<details>
 <summary>
   <code>/etc/logid.cfg</code>
 </summary>

```libconfig
devices: (
{
    name: "MX Master 3";

    buttons: (
        {
            cid: 0x50;
            action =
            {
                type: "None";
            };
        },
        {
            cid: 0x51;
            action =
            {
                type: "None";
            };
        },
        {
            cid: 0x52;
            action =
            {
                type: "None";
            };
        },
        {
            cid: 0x53;
            action =
            {
                type: "None";
[... 1185 lines cut ...]
        {
            cid: 0xfd;
            action =
            {
                type: "None";
            };
        },
        {
            cid: 0xfe;
            action =
            {
                type: "None";
            };
        }
    );
}
);
```

_Note: try `type: "Keypress";` with `keys: ["KEY_F24"];` if `"None"` doesn't work!_
</details>

As I said, I can't test it right now, so I'm not even sure this will work, but at least it'll give you a headstart. With it, start `logid` and check the warnings - all the valid CIDs are the codes in the config that have **not** thrown warnings.

To be fair, not the most elegant of solutions, but worked for me back then.