PixlOne/logiops #279: logid only works after restarting service

epassaro, 2021-11-23:
I never experienced #204 until a recent openSUSE Tumbleweed update (like a month ago).

Installed the latest version https://github.com/PixlOne/logiops/commit/6bb470000952405b6817032c43c7215f7d070428 and my `logid.service` looks ok:

```
### Editing /etc/systemd/system/logid.service.d/override.conf
### Anything between here and the comment below will become the new contents of the file

### Lines below this comment will be discarded

### /usr/lib/systemd/system/logid.service
# [Unit]
# Description=Logitech Configuration Daemon
# StartLimitIntervalSec=0
# After=multi-user.target
# Wants=multi-user.target
# 
# [Service]
# Type=simple
# ExecStart=/usr/local/bin/logid
# User=root
# ExecReload=/bin/kill -HUP $MAINPID
# Restart=on-failure
# 
# [Install]
# WantedBy=graphical.target
```

but still need to restart the service after login. Any ideas?

DominikMendel, 2021-12-12:
I also get this issue every startup without fault. I have verified the service is enabled and running. It just needs to be restarted to properly work. 
A fix for this would be great.

liznevada, 2021-12-27:
can confirm.
additionally this is what the unit returns on querying its status right after login
`○ logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; disabled; vendor preset: disabled)
     Active: inactive (dead)
`

epassaro, 2021-12-28:
It was fixed automatically for me. Now after some minutes, `logiops` starts working.

Staindk, 2022-01-13:
Been having the same issue for months (running Gnome on Arch).

Realised recently that instead of restarting `logid` I could just switch my mouse off and on and it will work :shrug:  thought it may be worth mentioning here as it is much easier than manually restarting the service.

Previously I had tried amending `logid.service` as the OP stated (and with some other tweaks too) and none of those changes seemed to fix this issue.

(MX Anywhere 3 connected via bluetooth)

epassaro, 2022-01-13:
> Been having the same issue for months (running Gnome on Arch).
>
> Realised recently that instead of restarting `logid` I could just switch my mouse off and on and it will work shrug thought it may be [...]

Hi! Switching mouse on and off worked for me. Thanks for the tip!

JohanAke1, 2022-07-22:
What has helped me a bit is to add delay to the service start like
`ExecStartPre=/bin/sleep 30`
But that is only a band aid. I would guess that the failure comes from that the mice is not available when the service starts so it can't connect to it. So if one moves the mice before, it will become active and the service finds it correctly.

Better solution would be to kill the logid when it gets the error for "Invalid Function ID" or similar so Systemd would then restart the service and the mouse would have time to become active.

hez01, 2022-11-27:
For me, this happened when I used `--now` switch when setting the service to `enabled` (`systemctl enable --now logid`). This issue disappeared when I re-set it without that switch.

dszmaj, 2022-12-13:
I had this issue on Fedora 37 and @hez01's hint fixed it completely for me. I just restarted computerand everything is working fine. Thanks!

joaquinvacas, 2024-02-06:
This is still happening.

Is there any solution for this at this moment?

Anyhow, I may look forward to get something like an udev rule that may get the work done (at least to restart the service).

lazarosgogos, 2024-04-06:
This issue popped up again, the fix as described in #148 (using a udev rule) and #269 (using modprobe) do not work. Up until yesterday it all worked silky smooth. After performing a DNF package update suddenly it seems the service must be  restarted each time the mouse is turned on.

Any insights may prove useful.

lazarosgogos, 2024-04-07:
The latest [bluez](https://github.com/bluez/bluez/issues/791) update (5.73-3) seems to have messed with it as reported by other people as well. A downgrade fixes it for now, it seems to be irrelevant to the driver itself. Hope this helps someone.

wsilva98, 2024-11-12:
Hello any updates? Get this too (@liznevada ):

```bash 
○ logid.service - Logitech Configuration Daemon
     Loaded: loaded (/usr/lib/systemd/system/logid.service; disabled; preset: enabled)
     Active: inactive (dead)
```

For some reason even in the journal logs It doesn't even try to start the service...