PixlOne/logiops #280: Logid stops working, service restart needed

MorganMLGman, 2021-11-24:
After a reboot logid works for few minutes and then stops working. Service restart is needed with  `sudo systemctl restart logid`. `sudo systemctl status logid` gives no information 
[image]

tomekand1, 2021-12-02:
is that happening after sleep mode activated ??

MorganMLGman, 2021-12-03:
No, no sleep mode. Just using my keyboard and mouse normalny and after like 10 minutes my gesture button on mouse and function keys on keyboard stops working properly

VDuchauffour, 2022-02-01:
Same here when my MX Master 3 is deactivated after 10 minutes.

GarThor, 2025-03-03:
I am also having this issue.  I don't know how long it is before I restart it, and when it stops working though... usually it happens after I've stepped away from my computer for a period of time.

I'm not sure if there are logs somewhere that would help diagnose this issue?  Maybe my mouse goes to sleep, and logid can't detect it when it comes back online?