PixlOne/logiops #281: Scrolling not working at Gestures mode

Fish-Who, 2021-11-29:
Hello,
I am trying to swtich central scroll to horizontal scroll when gesture button is pressed.
However, with this configuration, the action or axis at gesture mode with direction `ScrollDown` or `ScrollUp` does not work, and I just couldn't figure out why.
Please point out if there is something wrong in the cfg or if it is a bug, thank you.

`devices:
(
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 16;
    };
	hiresscroll:
    {
       hires: true;
       invert: false;
       target: true;
       up:
       {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 2;
        },
        down:
        {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -2;
        },
    };
    dpi: 1000;

    buttons:
    (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures:
                (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F8"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_D"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_RIGHT"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_LEFT"];
                        };
                    },
                    {
                        direction: "None";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F9"];
                        };
                    },
                    {
                        direction: "ScrollDown";
                        threshold: 1;
                        mode: "Axis";
                        axis: "REL_HWHEEL_HI_RES";
                        axis_multiplier: -2;
                    },
                    {
                        direction: "ScrollUp";
                        threshold: 1;
                        mode: "Axis";
                        axis: "REL_HWHEEL_HI_RES";
                        axis_multiplier: 2;
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "Keypress";
                keys: ["KEY_PLAYPAUSE"];
            };
        },
        {
            cid: 0x56;
            action =
            {
                type: "Keypress";
                keys: ["KEY_LEFTMETA", "KEY_LEFTSHIFT", "KEY_PRINT"];
            };
        },
        {
            cid: 0x53;
            action =
            {
                type: "Keypress";
                keys: ["KEY_LEFTCTRL"];
            };
        }
    );
}
);
`

S-trace, 2021-12-06:
Please take a look at https://github.com/PixlOne/logiops/pull/245 pull request - I also faced this bug and fixed it.

Fish-Who, 2021-12-11:
Thanks for replying, but it seems that the issue you had have been solved in the lastest release, and my mouse can perform `REL_HWHELL` in gesture mode without modifing the source code, my issue instead is that the `ScrollUp` or `ScrollDown` can't function in gesture mode, regardless the mode is onrelease or axis, and whatever the action is.

Fish-Who, 2021-12-11:
Now solved by using the branch from pull request #161