PixlOne/logiops #285: Scrolling is still very sensitive after upgrading `xf86-input-libinput` to 1.2.0

Pistrie, 2021-12-22:
The issue described in #271 still exists. This is an application bug according to [this thread](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/-/issues/39). Is there something that can be done without downgrading the package?

Ashark, 2022-01-23:
Just wanted to mention my workaround when I did not write hires statement in logid.cfg for my MX Master.
When I started logid and also when move mouse after a while (from powersave mode), the scrolling became unusably slow.
To workaround this, I launched solaar, and ticked "Resolution of mouse wheel"  two times (off, then on). And scrolling became normal, until next waking from mouse sleep.

But now when I used `    hiresscroll: { hires: true; invert: false; target: false; };` in config, scrolling is normally fast. It is on kde wayland.

johnhamelink, 2022-02-25:
Hey friends, there is now an option to fix this issue with libinput 1.2.x:

> The steps I took were to copy the default libinput Xorg config (/usr/share/X11/xorg.conf.d/40-libinput.conf) to /etc/X11/xorg.conf.d/, then to edit the 'pointer catchall' section to include Option "HighResolutionWheelScrolling" "off".

lferra, 2022-06-27:
> Hey friends, there is now an Xorg option to fix this issue with libinput 1.2.x:
> 
> > The steps I took were to copy the default libinput Xorg config (/usr/share/X11/xorg.conf.d/40-libinput.conf) to /etc/X11/xorg.conf.d/, then to edit the 'pointer catchall' section to include Option "HighResolutionWheelScrolling" "off".
> 
> My 40-libinput.conf looks like this now:
> 
> ```
> Section "InputClass"
>         Identifier "libinput pointer catchall"
>         MatchIsPointer "on"
>         MatchDevicePath "/dev/input/event*"
>         Option "HighResolutionWheelScrolling" "off"
>         Driver "libinput"
> EndSection
> 
> Section "InputClass"
>         Identifier "libinput keyboard catchall"
>         MatchIsKeyboard "on"
>         MatchDevicePath "/dev/input/event*"
>         Driver "libinput"
> EndSection
> 
> Section "InputClass"
>         Identifier "libinput touchpad catchall"
>         MatchIsTouchpad "on"
>         MatchDevicePath "/dev/input/event*"
>         Driver "libinput"
> EndSection
> 
> Section "InputClass"
>         Identifier "libinput touchscreen catchall"
>         MatchIsTouchscreen "on"
>         MatchDevicePath "/dev/input/event*"
>         Driver "libinput"
> EndSection
> 
> Section "InputClass"
>         Identifier "libinput tablet catchall"
>         MatchIsTablet "on"
>         MatchDevicePath "/dev/input/event*"
>         Driver "libinput"
> EndSection
> ```

Brilliant! This also fixed an odd behaviour I had with latte-dock with the cycle through tasks function. It always worked fine and probably the issue started with the xf68-input-libinput 1.2x upgrade.

Thanks

closed 2023-02-16 by Pistrie