PixlOne/logiops #287: Is there something like `action: "execute"`?

matthis-k, 2022-01-05:
I currently use a MxMaster 3 and want the up-gesture to open my application list, which is a script.
Is there something that enables me to write something similar to this:
```
gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Execute";
                            command: "myapplauncher";
                        };
                   }
);
```

PixlOne, 2022-01-30:
This may be a feature in the future, but right now there are some issues with this. The daemon is designed to be run as root and as a service, and therefore does not have access to the desktop. Also, keybinds should not run as root.

After I finish the IPC feature and add a GUI for configuring this, I plan to also create a desktop helper service. This will run as the user and will pave the way for certain features including:
 - Switching profiles based on the currently focused app.
 - Logitech Flow support
 - Proper app execution support.

No guarantees about any of those though, especially if you're using Wayland.

johantiden, 2022-03-28:
As a workaround, I'd suggest mapping it to a key and then mapping a global shortcut in your distro config to that script.

I think this is a better solution overall because it decouples the input from the action. It should be up to the desktop to map from input to action.

maxmax91, 2023-02-24:
I'm also waiting for this feature.
The desired behaviour I want to obtain:

WHEN I click with the central button of the mouse
THEN both keyboard and mouse switch to the next host as action type: "ChangeHost"; does.

I would be able to do it if only I could execute a shell program with the key or gesture.

Since I have a windows machine and a linux machine, if I manage to replicate the same behaviour on Windows, then the result will be really similar (if not better) to logitech flow.

Ashark, 2023-02-27:
> if I manage to replicate the same behaviour on Windows, then the result will be really similar (if not better) to logitech flow.

Is it possible to use such fictions in Windows bypassing logitech official soft? I am also interested in this, because I want to make a "flow mode" in a barrier/synergy/input leap. This will allow you to switch between hosts not in one network.