PixlOne/logiops #288: Issues mapping horizontal scroll wheel on MX Master 2S

2Manchu, 2022-01-07:
I'm having issues getting my horizontal scroll wheel to work, and apologies if there's something obvious I'm missing since I'm still a relatively new Linux user. From what I can tell, my mouse _should_ support horizontal scroll remapping with this project since it uses HID++ 2.0 (I am using the original included Unifying receiver), but all the configuration I've tried to do has not worked. This is my `logid.cfg`:

```
devices: (
{
    name: "Wireless Mouse MX Master 2S";
    ...
    ///smartshift, hires scroll config, omitted here for brevity
    ...
    thumbwheel:
    {
	divert: true;
	invert: false;
	left:
	    {
		direction: "Left";
		mode: "OnInterval";
		interval: 50;
		action = 
		    {
			type: "Keypress";
			keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
		    };
	    };
	right:
	    {
		direction: "Right";
		mode: "OnInterval";
		interval: 50;
		action = 
		    {
			type: "Keypress";
			keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
		    };
	    };		
    };
    buttons: (
        {
            ...
            //buttons config, again omitted
            ...
        }
    );
}
);

```

From what I've read on the other issues, I should be getting a readout regarding the thumbwheel when I run `sudo logid -v DEBUG`, however I do not, and I only get the following:

```
...
[INFO] Detected receiver at /dev/hidraw1
[DEBUG] Unsupported device /dev/hidraw6 ignored
[INFO] Device found: Wireless Mouse MX Master 2S on /dev/hidraw1:1
[DEBUG] /dev/hidraw1:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[WARN] Error adding device /dev/hidraw5: std::exception
```

I have checked to make sure that I compiled using the latest commit on this repository as suggested in another issue, as well as swapped USB ports on and off of the hub on my monitor. Thanks for all the great work on this project!

rstanuwijaya, 2022-06-05:
I am having the exact same issue. Also using MX master 2s.

jacob-mack, 2022-09-29:
Did you ever find a way to fix the horizontal scroll issue?  I seem to be struggling with the same thing.

Phen-Ro, 2023-05-27:
Try dropping the interval to 1, as mentioned in #145.  That's what worked for me.