PixlOne/logiops #295: Solaar + logiops?

MicaelJarniac, 2022-02-03:
I've just got a Logitech MX Master 3, and I've been configuring it with Solaar.

Most of the things work perfectly fine, but there are a couple of features missing, like the gesture actions. However, it seems that logiops does support the gestures.

I was wondering if I can use Solaar for configuring most things, and logiops just for the gestures, or if I have to pick one or the other, and how I should go about it.

I also have a Logitech G915 that I use with Solaar.

I like Solaar's interface, and would really like to keep using it for configuring the mouse, and only use logiops for things that Solaar doesn't support yet.

MicaelJarniac, 2022-02-03:
Apparently, Solaar does have some support for mouse gestures, but you have to use the "rules editor".

[image]

I'm not sure if it's as detailed as logiops' gestures, though.

pfps, 2022-05-16:
You could directly edit the Solaar rules file in ~/.config/solaar/rules.yaml but there is little documentation on the file format.