PixlOne/logiops #296: SmartShift setting is not updated on MX Master 3 after mouse reboot - SmartShift toggle needed

S-trace, 2022-02-11:
Hello.

I have the following section in my logid.cfg:
```
    smartshift: { 
        on: true; 
        threshold: 10;
        default_threshold: 10;
    };
```
It speeds up SmartShift triggering to make it easier.
But after the mouse reboot, this setting is not uploaded to the mouse, and I have to toggle SmartShift off and then back on using the SmartShift hardware button on the mouse to get this setting applied.

Is it possible to upload this setting to the mouse right after connection without any manual interaction?

Thank you.

nicochunger, 2022-05-30:
I believe you need to run `sudo logid` for the settings to be applied to the mouse

S-trace, 2022-07-04:
> I believe you need to run `sudo logid` for the settings to be applied to the mouse

I have logid launch on boot in systemd unit - so, it is started on boot.