PixlOne/logiops #300: No DJ reports for MX Master 3

pnou, 2022-03-02:
I have just bought a MX Master 3 mouse and it acts like it's not supported. With the last git version of logiops, I get the following error (it's only a warning but I guess it prevents the whole thing to work):

dimitri@corwin:~/logiops/build$ ./logid -vV
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[INFO] Detected receiver at /dev/hidraw2
[WARN] Error adding device /dev/hidraw2: No DJ reports

Here are some debugging information:

dimitri@corwin:~/logiops/build$ lsusb | grep Receiver
Bus 001 Device 007: ID 046d:c548 Logitech, Inc. USB Receiver

dimitri@corwin:~/logiops/build$ sudo usbhid-dump -d 046d:c548              
001:007:002:DESCRIPTOR         1646239789.856274
 06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
 A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81                                        
 00 09 02 91 00 C0

001:007:001:DESCRIPTOR         1646239789.857870
 05 01 09 02 A1 01 85 02 09 01 A1 00 95 10 75 01
 15 00 25 01 05 09 19 01 29 10 81 02 95 02 75 10
 16 01 80 26 FF 7F 05 01 09 30 09 31 81 06 95 01
 75 08 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38
 02 81 06 C0 C0 05 0C 09 01 A1 01 85 03 95 02 75
 10 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
 09 80 A1 01 85 04 95 01 75 02 15 01 25 03 09 82
 09 81 09 83 81 00 75 01 15 00 25 01 09 9B 81 06
 75 05 81 03 C0

001:007:000:DESCRIPTOR         1646239789.860069
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 05 05 08 19 01 29 05 91 02
 95 01 75 03 91 03 95 70 75 01 05 07 19 04 29 73
 81 02 95 05 19 87 29 8B 81 02 95 03 19 90 29 92
 81 02 C0

Do you understand what happens?

pnou, 2022-03-02:
I would be happy to help you in any way. I'm a developer, this can make things easier.

For the moment, I will buy a Bluetooth receiver.

Thanks for your help.

fftux, 2022-03-16:
Happy to help with testing here.  I just picked up an MX Keys Mini for Business and an MX Master 3 for Business.  I am using both via bluetooth at the moment, but plan to use the bolt receiver in the future.

atombombzero, 2022-08-09:
I was able to get the bolt dongle to connect my mouse (MX Master 3S), though it can be laggy. I was curious about getting support for it in the app.
```
root@arm-ubuntu2004:~# hostnamectl 
   Static hostname: arm-ubuntu2004
   Pretty hostname: arm-ubuntu20_04
         Icon name: computer-laptop
           Chassis: laptop
        Machine ID: c26f23b44f20474d8f6a35fe411b0c4c
           Boot ID: 609050293e594547b77458078ccfa078
  Operating System: Ubuntu 20.04.4 LTS
            Kernel: Linux 5.14.0-1047-oem
      Architecture: x86-64
root@arm-ubuntu2004:~# logid --version
v0.2.2-70-gdbe2b28

Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc548 
  bcdDevice            5.01
  iManufacturer           1 Logitech
  iProduct                2 USB Receiver
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 MPR05.01_B0010
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     133
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      54
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

FSund, 2022-08-24:
I have a "MX Master 3 for Business" that comes with a Logi Bolt adapter, and would be willing to help with implementing support for it. Any hints to where I should start?

pranaypratyush, 2022-08-28:
@atombombzero I too have the Master 3S. Were you able to get the bolt working?

e-s-a-i, 2022-12-02:
Another new owner reporting in, are there any developments on this issue? 
Is there any info we could gather or could someone point to a stating point for getting it working?

aaronbjohnson, 2022-12-06:
Anybody make any progress with this yet? Haven't been able to make any headway myself.

amitrajput1992, 2022-12-20:
I was able to make it work via Bluetooth but not the bolt connector. I have been using it for the past couple of days and haven't faced any lag via Bluetooth. Here's my config if anyone's interested (pretty minimalist) https://github.com/amitrajput1992/logiops/blob/master/logid.amit.cfg

I am using TPLink UB500, with Linux mint kernel 5.15.0-56-generic

I'm more than happy to help with any testing.

TasPats, 2023-01-15:
how about get some Logitech Bolt code from Solaar project ( https://github.com/pwr-Solaar/Solaar ) or it is not possible?

hliu202, 2023-01-17:
I worked around this by using **Solaar** (`sudo apt install solaar` on ubuntu). Now the DPI is configured to 2000.

I can't believe the default DPI is 1000, which is too small for 4K displayer!

CharlesChiuGit, 2023-01-17:
A MX master 3S user here, connected via Bolt. Now use solaar as a workaround, hope the support come soon!

aaronbjohnson, 2023-01-17:
> A MX master 3S user here, connected via Bolt. Using solaar as a workaround for now(not a gui fan and I'm connecting the mouse via a kvm), hope the bolt support come soon!

Thank you, I had not heard of the solaar alternative. Are you able to disable mouse acceleration with it?

CharlesChiuGit, 2023-01-18:
> Thank you, I had not heard of the solaar alternative. Are you able to disable mouse acceleration with it?

Not sure about that, solaar is pretty basic. You can change the DPI tho.

nootums, 2023-03-25:
Yeah, the solaar app does not allow me to customize the buttons(M650 signature). For now I'm using Bluetooth to connect to the mouse, but the connection is choppy at times.

nootums, 2023-03-25:
> Not sure about that, solaar is pretty basic. You can change the DPI tho.

Yeah, The solaar app is very limited, I cannot customize the side buttons on my mouse(M650 Signature). For the time being, I'm using bluetooth, which is very choppy at times.

PixlOne, 2023-04-30:
Added Logi Bolt support in a branch that also adds preliminary GUI support and many other I/O fixes, please try #362.

closed 2023-05-11 by PixlOne

CharlesChiuGit, 2023-05-19:
This works fine even over KVM! Thanks!