PixlOne/logiops #301: Add MX Keys Mini to Tested Devices

Minionflo, 2022-03-04:
I have a Logitech MX Keys mini and it works just fine.

Minionflo, 2022-03-04:
I found out that logiops cant handle cids with a lenght over 2. For example 0xe9 works but 0x11c does not, i get the error `MX Keys Mini: CID 0x1c does not exist`

fftux, 2022-03-16:
@Minionflo would you care to rename this issue to reflect that the device has issues, rather than that it should be added to the tested devices list?  I had a similar issue open outlining the same issue that you have noted.  I close my issue noting it was a duplicate.

kamer, 2022-05-29:
I have the same problem with my MX Keys Mini.

Output of `logid -v`:
```
[INFO] Device found: MX Keys Mini on /dev/hidraw3:255
[DEBUG] /dev/hidraw3:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x34 |         |         |            | 
[DEBUG] 0xd1 |         | YES     |            | 
[DEBUG] 0xd2 |         | YES     |            | 
[DEBUG] 0xd3 |         | YES     |            | 
[DEBUG] 0xde |         | YES     |            | 
[DEBUG] 0xe2 | YES     | YES     |            | 
[DEBUG] 0xe3 | YES     | YES     |            | 
[DEBUG] 0xe5 | YES     | YES     |            | 
[DEBUG] 0xe7 | YES     | YES     |            | 
[DEBUG] 0xe8 | YES     | YES     |            | 
[DEBUG] 0xe9 | YES     |         |            | 
[DEBUG] 0x103 | YES     | YES     |            | 
[DEBUG] 0x108 | YES     | YES     |            | 
[DEBUG] 0x10a | YES     | YES     |            | 
[DEBUG] 0x117 | YES     |         |            | 
[DEBUG] 0x11c | YES     | YES     |            | 
```

I can see 0x11c in the remapple list. But following config does not work: 
```
{
    cid: 0x11C; 
    action = {
      type: "Keypress";
      keys: ["KEY_MICMUTE"];
    };
  }
```

`0x11c` is recognized as `0x1c`.
```
[WARN] MX Keys Mini: CID 0x1c does not exist.
```
----
I'm ready to test or provide information about device to contribute. @PixlOne