PixlOne/logiops #307: No way to opt out of systemd service installation

ckiee, 2022-03-27:
Hey! I'm packaging this project for [NixOS](https://nixos.org) because I just got an MX Master 3 and currently it's failing because NixOS makes already-built packages read-only:

```
[100%] Built target logid
Install the project...
-- Install configuration: "Release"
-- Installing: /nix/store/q9gi9bq704hf9m94v9sahma3lg49pgvp-logiops-0.2.3/bin/logid
CMake Error at src/logid/cmake_install.cmake:70 (file):
  file cannot create directory:
  /nix/store/kxqqbyxf4w0bg4n2ip1qq3kr5bw4hdq0-systemd-249.7/lib/systemd/system.
  Maybe need administrative privileges.
Call Stack (most recent call first):
  cmake_install.cmake:47 (include)
```

The relevant [CMake file](https://github.com/PixlOne/logiops/blob/dbe2b28a5307dfc25e7b4858b94bded9acc5ea5c/src/logid/CMakeLists.txt#L92) does not have a case to allow me to set `SYSTEMD_SERVICES_INSTALL_DIR` to another directory, so I'll have to patch it. Can you add a case that lets the user specify the `SYSTEMD_SERVICES_INSTALL_DIR`?

anund, 2022-07-29:
@ckiee setting `PKG_CONFIG_SYSTEMD_SYSTEMDSYSTEMUNITDIR` worked. Close this issue?

ckiee, 2022-07-29:
Oh yeah, I forgot I made it. Sorry!

closed 2022-07-29 by ckiee