PixlOne/logiops #313: MX master 3 scroll wheel is hard to scroll

sash-a, 2022-05-03:
When trying to scroll the scroll wheel is incredibly tight and hard to move (especially for scroll up). Is there an option to make it easier to scroll?

`threshold`, `default_threshold` and `axis_multiplier` all seem to have no effect.

Here's my logid.cfg

```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 10;
        default_threshold: 10
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
        up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 3;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -3;
        },
    }
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "None"
                        action = 
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA"]

                        }
                    }
                );
            };
        },
        # {
        #     cid: 0xc4;
        #     action =
        #     {
        #         type: "Keypress";
        #         keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_P"];
        #     };
        # }
    );
}
);
```

Litosty, 2022-08-15:
Try this:
```
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: true;
        up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 3;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -3;
        },
```

Tolgon, 2022-08-26:
I've been trying the same thing. I've set target to true, applied the same up/down configuration but no cigar. The scroll speed is still not working great on Wayland Gnome (libinput).