PixlOne/logiops #320: MX vertical mouse gesture action doesn't work

joyet-simon, 2022-05-19:
Hi, 
I have no errors but my config for gestures on back button doesn't work. This is ma config:
```
devices: (
{
    name: "MX Vertical Advanced Ergonomic Mouse";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1200;

    buttons: (
        {
            cid: 0x53;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_COPY"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_PASTE"];
                        };
                    },
                    {
                        direction: "None"
                        mode: "NoPress"
                    }
                );
            };
        }
    );
}
);
```

I juste wan't a copy and a paste action (for the moment).  Anyone can tell me what is not good in my file?

LeonMatthesKDAB, 2022-10-14:
If you're trying to use the "DPI switch button" at the top of the mouse, you're using the wrong cid.
The MX Vertical is strange in that regard.
The correct cid is: 0xfd, not 0x53.

joyet-simon, 2022-10-14:
> If you're trying to use the "DPI switch button" at the top of the mouse, you're using the wrong cid. The MX Vertical is strange in that regard. The correct cid is: 0xfd, not 0x53.

No, I try to use the "backward button" and the "forward button". The two lateral button of the mouse. (thanks for your reply)

nottoday888, 2022-10-14:
Have you tried only 0x53? The [wiki](https://github.com/PixlOne/logiops/wiki/CIDs) lists four different IDs for the back button. Namely: 0x0053, 0x0054, 0x00bd and 0x00ce.

joyet-simon, 2022-10-14:
> Have you tried only 0x53? The [wiki](https://github.com/PixlOne/logiops/wiki/CIDs) lists four different IDs for the back button. Namely: 0x0053, 0x0054, 0x00bd and 0x00ce.

I tried this just now and it doesn't work.

wnykuang, 2023-04-11:
Have you tried start from a scratch config file with only key settings? I don't think 

```
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };

```
is supported by vertical, I just found the issue. 

And have you tried to run ./logiops with -v?

gkoma, 2024-03-17:
The configuration that I use for the Logitech MX Vertical, the top button is used for the window snapping gesture.

`// Logiops (Linux driver) configuration for Logitech MX Vertical Advanced Ergonomic Mouse
// Includes gestures
// Tested on logid v0.2.2-59-g6bb4700 - GNOME Shell 42.9 on Ubuntu 22.04.4 LTS
// Whats working:
//   1. Window snapping using Gesture button (Top button)
//   2. Forward Back Buttons

// File location: /etc/logid.cfg

devices: ({
  name: "MX Vertical Advanced Ergonomic Mouse";

  smartshift: {
    on: true;
    threshold: 15;
  };

  hiresscroll: {
    hires: true;
    invert: false;
    target: false;
  };

  dpi: 1500; // max=4000

  buttons: (
    // Forward button
    {
      cid: 0x56;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_FORWARD" ];
            }
          }
        );
      };
    },
    // Back button
    {
      cid: 0x53;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_BACK" ];
            }
          }
        );
      };
    },

    // Gesture button (hold and move)
    {
      cid: 0xfd;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA" ]; // open activities overview
            }
          },

          {
            direction: "Right";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA", "KEY_RIGHT" ]; // snap window to right
            }
          },

          {
            direction: "Left";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA", "KEY_LEFT" ];
            }
		  },

		  {
            direction: "Up";
            mode: "onRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA", "KEY_UP" ]; // maximize window
            }
		  },
		  
		  {
            direction: "Down";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA", "KEY_DOWN" ]; // minimize window
            }
          }
        );
      };
    }
  );
});`

prashantgupta7775, 2024-03-21:
Can you share the root location of etc/logid.cfg as I am getting error.
[INFO] Config file does not exist, using empty config.
[INFO] Detected receiver at /dev/hidraw5

rautamiekka, 2024-03-21:
> Can you share the root location of etc/logid.cfg as I am getting error. [INFO] Config file does not exist, using empty config. [INFO] Detected receiver at /dev/hidraw5

`/etc/logid.cfg` is literal in its meaning; `/etc/logid.cfg` ain't the same as `etc/logid.cfg`.

prashantgupta7775, 2024-03-21:
Got it , config is detecting, but there is no change using it.
so there is any thing else need to do except sudo logid.

good-epic, 2024-05-23:
> The configuration that I use for the Logitech MX Vertical, the top button is used for the window snapping gesture.
>
> ```
> // Logiops (Linux driver) configuration for Logitech MX Vertical Advanced [...]

I tried this in Pop OS and it almost got it working. For me I needed to add:

```
    // Left button                                                              
    {
      cid: 0x50;
      action = {
        type: "Keypress";
        keys: [ "BTN_LEFT" ];
      };
    },
    // Right button                                                             
    {
      cid: 0x51;
      action = {
        type: "Keypress";
        keys: [ "BTN_RIGHT" ];
      };
    },
```

Just putting this here in case it helps anyone else.