PixlOne/logiops #321: Random errors

ivanpepelko, 2022-05-26:


akamensky, 2022-07-20:
Same here on Fedora 36:

```
[root@hostname:~] journalctl -feu logid
...
Jul 20 15:34:00 hostname.domain.tld logid[14358]: [WARN] Error adding device /dev/hidraw6: std::exception
...
Jul 20 15:34:43 hostname.domain.tld logid[14452]: [WARN] Error adding device /dev/hidraw6: std::exception
...
Jul 20 15:35:32 hostname.domain.tld logid[14511]: [WARN] Error adding device /dev/hidraw6: std::exception
...
Jul 20 15:37:52 hostname.domain.tld logid[14766]: [WARN] Error adding device /dev/hidraw6: std::exception
Jul 20 15:39:50 hostname.domain.tld logid[14766]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw4:1: Resource error
...
Jul 20 15:40:19 hostname.domain.tld logid[15090]: [WARN] Error adding device /dev/hidraw6: std::exception
Jul 20 16:15:07 hostname.domain.tld logid[15090]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw4:1: Resource error
Jul 20 17:05:07 hostname.domain.tld logid[15090]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw4:1: Resource error
Jul 20 17:07:29 hostname.domain.tld logid[15090]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw4:1: Resource error
Jul 20 17:10:54 hostname.domain.tld logid[15090]: [ERROR] Caught HID++ 1.0 error while trying to initialize /dev/hidraw4:1: Resource error
```
`WARN` ones don't seem to cause any issue, but `ERROR` corresponds to mouse freezing and acting very choppy.

Configuration:

```
[root@hostname:~] cat /etc/logid.cfg
devices: (
{
    name: "Wireless Mouse MX Master 3";
    dpi: 1200;
    buttons: (
        {
            cid: 0x0056;
            action =
            {
                type: "Keypress";
                keys: ["KEY_VOLUMEUP"];
            };
        },
        {
            cid: 0x0053;
            action =
            {
                type: "Keypress";
                keys: ["KEY_VOLUMEDOWN"];
            };
        },
        {
            cid: 0x00c3;
            action =
            {
                type: "None";
            };
        },
    );
}
);
```

Version running:

```
[root@hostname:~] rpm -qa | grep logiops
logiops-0.2.3^1.gitdbe2b28-10.fc36.x86_64
```

PixlOne, 2023-05-03:
The issue is that your connection to the mouse is poor. This manifests both as your cursor being choppy and logid failing to communicate with your device. I've recently made some changes that improve connection stability, the random errors should hopefully be fixed but the mouse choppiness is an issue with your machine.