PixlOne/logiops #323: Trying to connect a MX Master 1, please help

schickung, 2022-06-12:
Hi there, could you please help me with this issue, Iḿ desperate. Trying to connect a MX Master 1. I am getting this error:

[ERROR] I/O Error while reading /etc/logid.cfg: FileIOException
[INFO] Detected receiver at /dev/hidraw1
[WARN] Error adding device /dev/hidraw2: std::exception
[WARN] Error ignored on detached thread/task: _readReport read failed: Input/output error
[INFO] Receiver on /dev/hidraw1 disconnected
[INFO] Detected receiver at /dev/hidraw1
[WARN] Error adding device /dev/hidraw1: Invalid sub ID
[WARN] Error adding device /dev/hidraw2: std::exception

Fishbowler, 2022-09-18:
If you're using a dongle, try switching to bluetooth

PixlOne, 2023-05-03:
Try again in the latest version, I should hopefully have fixed that.