PixlOne/logiops #328: Could not create input device: No such device

PLNech, 2022-08-30:
I'm trying to use a Logitech MX Vertical mouse. After setting up my cfg file following the best example I could find (from #320), I see the following error when running `logid -v`:
```
[ERROR] Could not create input device: No such device
```
The mouse is paired/connected/trusted via bluetooth:
```
~ bluetoothctl info EA:CC:B3:B9:23:FB

Device EA:CC:B3:B9:23:FB (random)
	Name: MX Vertical
	Alias: MX Vertical
	Appearance: 0x03c2
	Icon: input-mouse
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00010000-0000-1000-8000-011f2000046d)
	Modalias: usb:v046DpB020d0009
	Battery Percentage: 0x64 (100)
```

However nothing happens when I move it or click.

Any idea how to troubleshoot this situation?

markstos, 2022-11-02:
Might be a dupe of #249 or #204.

Aworldc, 2023-11-05:
for me, `logid` doesn't work but `sudo logid` does.