PixlOne/logiops #336: Mx Master 2s no sane scrolling option for 1 scroll per 1 mouse wheel click

GideonUng, 2022-10-21:
Scrolling is either continuous and very sensitive regardless if smart shift is on and scrolls multiple times before 1 mouse wheel click or it scrolls only 1 line every 5 or so mouse wheel clicks.
those options correspond to `hires` on and off in the `highresscroll` option
i dont care about any buttons or mappings or other useless features the only thing i need is for this mouse to scroll normally.

Sewdohe, 2022-11-20:
I agree with this, and hires's scrolling speed is nuts...to solve though I just disabled hi-res and the click-scroll works as expected

disasmwinnie, 2023-01-16:
Same here, but didn't see it as an issue. Why is ``hires: false;`` not an option respectively why is having HiRes on scroll-whell important? (Have not experience with it).

closed 2023-11-21 by GideonUng

lalten, 2023-12-01:
@GideonUng did you find a solution? Scroll speed is still way too high with my Master 2s on Arch / logid v0.3.3 regardless of `hires` being true or false.