PixlOne/logiops #339: MX Master 3S includes movement before button press in determining Gesture direction

austin5627, 2022-10-30:
When pressing a button with a gesture, movements taken before pressing the button affect the direction making it difficult to use the None direction

relevant config section
```
cid: 0xc3; # Gesture Button/Left Thumb button
action =
{
    type: "Gestures"
    gestures: (
        {
            direction: "None"; # Play/Pause
            mode: "OnRelease";
            action =
            {
                type: "Keypress";
                keys: ["KEY_PLAYPAUSE"];
            };
        },
        {
            direction: "Left"; # Desktop forward
            mode: "OnRelease";
            action =
            {
                type = "Keypress";
                keys: ["KEY_LEFTMETA", "KEY_COMMA"];
            };
        },
        {
            direction: "Right"; # Desktop backwards
            mode: "OnRelease";
            action =
            {
                type = "Keypress";
                keys: ["KEY_LEFTMETA", "KEY_DOT"];
            };
        },
    );
};

```

nottoday888, 2022-10-31:
Have you tried playing with the threshold option?
https://github.com/PixlOne/logiops/wiki/Configuration#threshold-1

austin5627, 2022-10-31:
Yes. The problem is that any movement that exceeds the threshold before pressing the button will trigger the gesture in that direction. Even if the mouse is completely stationary for 30 seconds before pressing the button, if it was moved to the left before that period, pressing the button will trigger the left action

ZhaoyangZh, 2022-11-08:
Hi, I am having trouble with using MX Master 3S too, do you mind sharing your logid.cfg? Many thanks!

austin5627, 2022-11-08:
Sure
```
# this config file is for Logiops and needs to be placed in /etc/logid.cfg
devices: (
{
    name: "MX Master 3S";
    smartshift:
    {
        on: true;
        threshold: 10;
        default_threshold: 10;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 1500;
    thumbwheel:
    {
        divert: true;
        invert: false;
        left =
        {
            direction: "Left";
            mode: "OnInterval";
            threshold: 1;
            interval: 10;
            action =
            {
                type: "Keypress";
                # keys: ["KEY_CAPSLOCK", "KEY_LEFTSHIFT", "KEY_TAB"];
                keys: ["KEY_LEFTMETA", "KEY_F3"];
            };
        }
        right =
        {
            direction: "Right";
            mode: "OnInterval";
            threshold: 10;
            interval: 10;
            action =
            {
                type: "Keypress";
                # keys: ["KEY_CAPSLOCK", "KEY_TAB"];
                keys: ["KEY_LEFTMETA", "KEY_F4"];
            };
        }
        tap =
        {
            type: "Keypress";
            keys: ["KEY_PLAYPAUSE"];
        }
    };

    buttons: (
        {
            cid: 0xc3; # Gesture Button/Left Thumb button
            action =
            {
                type: "Gestures"
                gestures: (
                    # {
                    #     direction: "None"; # Play/Pause
                    #     mode: "OnRelease";
                    #     action =
                    #     {
                    #         type: "Keypress";
                    #         keys: ["KEY_PLAYPAUSE"];
                    #         # keys: ["KEY_N", "KEY_O", "KEY_SPACE"]
                    #     };
                    # },
                    {
                        direction: "Left"; # Desktop forward
                        mode: "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_COMMA"];
                            # keys: ["KEY_L", "KEY_E", "KEY_F", "KEY_T", "KEY_SPACE"];
                        };
                    },
                    {
                        direction: "Right"; # Desktop backwards
                        mode: "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_DOT"];
                            # keys: ["KEY_R", "KEY_I", "KEY_G", "KEY_H", "KEY_T", "KEY_SPACE"];
                        };
                    },
                    # {
                    #     direction: "Down";
                    #     mode: "OnRelease";
                    #     action =
                    #     {
                    #         type: "Keypress";
                    #         keys: ["KEY_PLAYPAUSE"]
                    #     };
                    # },
                    # {
                    #     direction: "Up";
                    #     mode: "OnRelease";
                    #     action =
                    #     {
                    #         type: "Keypress";
                    #         keys: ["KEY_PLAYPAUSE"]
                    #     };
                    # }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "Gestures"
                gestures: (
                    {
                        threshold: 100;
                        direction: "Left"; # Window forward
                        mode: "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_K"];
                        };
                    },
                    {
                        threshold: 100;
                        direction: "Right"; # Window backwards
                        mode: "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_J"];
                        };
                    },
                    {
                        threshold: 100;
                        direction: "Down"; # Toggle floating
                        mode: "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_F"];
                        };
                    },
                    {
                        threshold: 100;
                        direction: "Up"; # Toggle Fullscreen
                        mode: "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFTSHIFT", "KEY_F"];
                        };
                    },
                    {
                        threshold: 100;
                        direction: "None"; # Toggle layout
                        mode: "OnRelease";
                        action =
                        {
                            type = "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_TAB"];
                        };
                    }
                );
            };
        }
    );
});
```

ZhaoyangZh, 2022-11-08:
> Sure
>
> ```
> # this config file is for Logiops and needs to be placed in /etc/logid.cfg
> devices: (
> {
>     name: "MX Master 3S";
>     smartshift:
>     {
>         on: true;
>         threshold: 10; [...]

Thanks! Actually after using this config, I am experiencing the same issue as yours. I temporally remove those gestures then.

obj-obj, 2022-11-30:
I am also having this issue with an MX Master 3S

EduPereyra, 2022-12-29:
I'm too experiencing this bug. Any news about it?

PixlOne, 2023-05-03:
Just fixed this in a recent commit by ignoring the first mouse movement event.

closed 2023-05-03 by PixlOne

EduPereyra, 2023-05-04:
Works perfectly. Thanks!

zhimsel, 2023-05-09:
For posterity, this appears to the be the commit: https://github.com/PixlOne/logiops/commit/5e436a2bdf4b36473e3a258bdd187a6fbfba7344