PixlOne/logiops #340: Thumb button gesture not working on MX Master 3S

ZhaoyangZh, 2022-11-08:
Hi there,

The thumb button gestures including opening activities overview are not working with my MX Master 3S on Ubuntu 22.04. 

```
[INFO] Device MX Master 3S not configured, using default config.
[INFO] Device found: MX Master 3S on /dev/hidraw1:255
[DEBUG] /dev/hidraw1:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```

How should I config my /etc/logid.cfg for MX 3S?

MarcioFCMartins, 2022-11-09:
I have the MX3, not 3s, and Pop_os 22.04, not Ubuntu. Still it should be similar. Here's my Thumb button gesture section:

```
// Thumb button
		{ cid: 0xc3; action = 
			{
                type: "Gestures";
                gestures: (
                    {
                        direction: "Left";
                        mode: "OnFewPixels";
                        pixels: 150;
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_W"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnFewPixels";
                        pixels: 150;
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_S"];
                        };
                    },
                    {
                        direction: "None";
                        mode: "OnRelease";
                        action = 
                        {
                        	type: "Keypress";
                        	keys: ["KEY_LEFTMETA"];
                        }
                    }
                );
            };
        },
```

ZhaoyangZh, 2022-11-12:
> I have the MX3, not 3s, and Pop_os 22.04, not Ubuntu. Still it should be similar. Here's my Thumb button gesture section:
>
> ```
> // Thumb button
> 		{ cid: 0xc3; action =
> 			{
>                 type: [...]

Thanks, I have solved the issue. The MX3S should use the device name "MX Master 3S" instead of "Wireless Mouse MX Master 3S".

u-ways, 2022-11-21:
Hello guys, I recently bought the 3S, can one of you share the full `logid.cfg` with me as I would be interested to see some 3S samples, please? :)

ZhaoyangZh, 2022-11-30:
> Hello guys, I recently bought the 3S, can one of you share the full `logid.cfg` with me as I would be interested to see some 3S samples, please? :)

```
// Logiops (Linux driver) configuration for Logitech MX Master 3.
// Includes gestures, smartshift, DPI.
// Tested on logid v0.2.3 - GNOME 3.38.4 on Zorin OS 16 Pro
// What's working:
//   1. Window snapping using Gesture button (Thumb)
//   2. Forward Back Buttons
//   3. Top button (Ratchet-Free wheel)
// What's not working:
//   1. Thumb scroll (H-scroll)
//   2. Scroll button

// File location: /etc/logid.cfg

devices: ({
  name: "MX Master 3S";

  smartshift: {
    on: true;
    threshold: 80;
  };

//  hiresscroll: {
//    hires: true;
//    invert: false;
//    target: false;
//  };
  hiresscroll:
  {
    hires: true;
    invert: false;
    target: true;
    up: {
        mode: "Axis";
        axis: "REL_WHEEL_HI_RES";
        axis_multiplier: 2;
    },
    down: {
        mode: "Axis";
        axis: "REL_WHEEL_HI_RES";
        axis_multiplier: -2;
    },
  }

  dpi: 2000; // max=4000

  buttons: (
    // Forward button
    {
      cid: 0x56;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_FORWARD" ];
            }
          },

          {
            direction: "Up";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_PLAYPAUSE" ];
            }
          },

          {
            direction: "Down";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA" ];
            }
          },

          {
            direction: "Right";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_NEXTSONG" ];
            }
          },

          {
            direction: "Left";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_PREVIOUSSONG" ];
            }
          }
        );
      };
    },

    // Back button
    {
      cid: 0x53;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_BACK" ];
            }
          }
        );
      };
    },

    // Gesture button (hold and move)
    {
      cid: 0xc3;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA" ]; // open activities overview
            }
          }

//          {
//            direction: "Right";
//            mode: "OnRelease";
//            action = {
//              type: "Keypress";
//              keys: [ "KEY_LEFTMETA", "KEY_RIGHT" ]; // snap window to right
//            }
//          },

//          {
//            direction: "Left";
//            mode: "OnRelease";
//            action = {
//              type: "Keypress";
//              keys: [ "KEY_LEFTMETA", "KEY_LEFT" ];
//            }
//	  },

//	  {
//            direction: "Up";
//            mode: "onRelease";
//            action = {
//              type: "Keypress";
//              keys: [ "KEY_LEFTMETA", "KEY_UP" ]; // maximize window
//            }
//	  }
        );
      };
    },
	
    // Top button
    {
      cid: 0xc4;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "ToggleSmartShift";
            }
          },

          {
            direction: "Up";
            mode: "OnRelease";
            action = {
              type: "ChangeDPI";
              inc: 1000,
            }
          },

          {
            direction: "Down";
            mode: "OnRelease";
            action = {
              type: "ChangeDPI";
              inc: -1000,
            }
          }
        );
      };
    }
  );
});

```

sviperz, 2024-07-12:
This is really weird. I've got new **MX Master 3S** to replace **MX Master 3** and now I have both of them on the table. Both are connected via BT. Both have the same config (see below) only with different device name. Thumb button actions are commented for both in the config. So, when I click the thumb button on **MX Master 3** it triggers app switch action on Ubuntu 24.04 as if I'd pressed META+TAB on the keyboard. But when I click the thumb button on **MX Master 3S** _nothing happens_. @PixlOne how is that possible? Do they have different handling for the thumb button? All other buttons work exactly the same way on both devices.

Here below is my config:
```
// Logiops Logitech MX Master 3 configuration.
// File location: /etc/logid.cfg

devices: ({
  name: "Wireless Mouse MX Master 3";

  smartshift: {
    on: true;
    threshold: 30; // 0 (free) - 50 (ratchet)
  };

  hiresscroll: {
    hires: true;
    invert: false;
    target: false;
  };

  dpi: 1000; // 200 (min) - 4000 (max)

  buttons: (
    // Forward Button
    {
      cid: 0x56;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTCTRL", "KEY_HOME" ]; // Scroll to top
            }
          },

          {
            direction: "Left";
            mode: "OnInterval";
            threshold: 1;
            interval: 100;
            action = {
              type: "Keypress";
              keys: [ "KEY_VOLUMEDOWN" ]; // Volume Down
            }
          },

          {
            direction: "Right";
            mode: "OnInterval";
            threshold: 1;
            interval: 100;
            action = {
              type: "Keypress";
              keys: [ "KEY_VOLUMEUP" ]; // Volume Up
            }
          },

          {
            direction: "Up";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTSHIFT", "KEY_HOME" ]; // Center window
            }
          },

[... 334 lines cut ...]

          {
            direction: "Up";
            mode: "OnRelease";
            action = {
              type: "CycleDPI";
              dpis: [1000],
            }
          },

          {
            direction: "Down";
            mode: "OnRelease";
            action = {
              type: "CycleDPI";
              dpis: [200],
            }
          }
        );
      };
    }
  );
}
);
```

sviperz, 2024-07-12:
I can add that by default the thumb button on **MX Master 3S** makes an action of focusing window without clicking inside that window.