PixlOne/logiops #342: MX Master 3S Configuration Reference

u-ways, 2022-11-22:
Hello guys, 

It took a bit of time to setup my MX Master 3S gestures using LogiOps, so here is a sample reference for anyone to see. As always, please have a look at the [Wiki's Configuration Page](https://github.com/PixlOne/logiops/wiki/Configuration) for further details.

**System Details**
- LogiOps version: `v0.2.3`
- Distribution: `Ubuntu 22.04.1 LTS`
- Kernel version: `Linux 5.15.0-53-generic x86_64 GNU/Linux`

<details>
<summary>My MX Master 3S <code>logid.cfg</code></summary>

```cfg
devices: ({
    name: "MX Master 3S";

    dpi: 2000;

    smartshift: {
        on: true;
        threshold: 30;
    };

    hiresscroll: {
        hires: false;
        invert: false;
        target: false;
        up: {
            mode: "Axis";
            axis: "REL_WHEEL";
            axis_multiplier: 1;
            },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL";
            axis_multiplier: -1;
        },
    };

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_UP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_DOWN"];
                        };
                    },
                    {
                        direction: "Left";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFT"];
                        };
                    },
                    {
                        direction: "Right";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_RIGHT"];
                        };
                    },
                    {
                        direction: "None";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA"];
                        }
                    }
                );
            };
        }
    );
})
```

<details/>

closed 2022-11-22 by u-ways

obj-obj, 2022-11-29:
You can also do this for the hi-res scroll:

```
    hiresscroll: {
        hires: false;
        invert: false;
        target: false;
        up: {
            mode: "Axis";
            axis: "REL_WHEEL";
            axis_multiplier: 1;
            },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL";
            axis_multiplier: -1;
        },
    };
```

julianosam, 2022-12-07:
I am trying your config but the gestures don't seem to trigger any action. Just to make sure I am doing it right, press and hold the thumb button and push the mouse to one of directions right? :sweat_smile:  I have a Master MX 3S as well. Thanks.

a1ad, 2023-11-26:
Did you guys got the thumbwheel scrolling working on the 3s?

obj-obj, 2023-11-29:
> Did you guys got the thumbwheel scrolling working on the 3s?

Yeah, you can make it high res scroll as well

obj-obj, 2023-11-30:
> Hmm can't seem to make it work, do you have a full config example for me?

I actually deleted my config a couple months ago since I stopped using my logitech mouse...