PixlOne/logiops #344: MX Master 3S - Unsupported Device

vanvince, 2022-12-01:
Hi, 

I'm trying to use my new mx master 3s on a virtual machine running Ubuntu.  The mouse is connected via bluetooth. 
Launching ```sudo logid -v``` this is the result:

``` [DEBUG] Unsupported device /dev/hidraw0 ignored  ``` 

This my logid.cfg file:
```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			   Keys:  ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_K"];

                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			   Keys:  ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_J"];

                        };
                    },

                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			   Keys:  ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_H"];

                        };
                    },

                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            Keys: ["KEY_LEFTMETA", "KEY_LEFTCTRL", "KEY_L"];

                        }
                    },

                    {
                        direction: "None"
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            Keys: ["KEY_LEFTMETA", "KEY_ENTER"];

                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                Keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                Keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
    );
},
{
# Another device to configure
name: "Other Logitech USB Receiver";

}
);
```
Do you know how to fix it ? 

Regards, 
Vincenzo

Leonidas90, 2022-12-02:
Try 
name: "MX Master 3S";

vanvince, 2022-12-02:
Hi @Leonidas90 ,

I tried but have the same issue.

bryankenote, 2022-12-03:
@aaronbjohnson 
Looks like your issue is the same as https://github.com/PixlOne/logiops/issues/300.
The Logi Bolt dongle is not supported yet. You can use bluetooth in the meantime.

Jenjen1324, 2022-12-14:
I just received my MX Master 3S and installed version 0.2.3-1 from AUR.
Using `name: "MX Master 3S";` worked without any issues when using bluetooth.

Note that as mentioned in the Arch wiki and #300, the new Bolt receiver the 3S uses is unsupported. So I opted to connect the mouse via Bluetooth.

Rami-Daoud, 2022-12-14:
> I just received my MX Master 3S and installed version 0.2.3-1 from AUR. Using `name: "MX Master 3S";` worked without any issues when using bluetooth.
> 
> Note that as mentioned in the Arch wiki and #300, the new Bolt receiver the 3S uses is unsupported. So I opted to connect the mouse via Bluetooth.

can you test name: "MX Master 3S B" and give feedback?

EduPereyra, 2022-12-28:
> can you test name: "MX Master 3S B" and give feedback?

I've tested with that name using the bolt receiver and it doesn´t work unfortunately.

PixlOne, 2023-05-03:
Logi Bolt is now supported in the latest release.

closed 2023-05-03 by PixlOne