PixlOne/logiops #348: Abandoned?

oscarpax, 2022-12-13:
Is logiops still actively developed/maintained?

chomosuke, 2022-12-28:
I have the same question as well. It seems that the maintainer hasn't responded to anything since feb 2022.

If @PixlOne has lost interest in this project, I wouldn't mind taking over tbh. I had already made some improvement to this project with #343 and I'd be interested to tackle #104 as well.

noahehall, 2022-12-28:
it definitely still works, recently used it to setup a logi max mechanical on ubuntu

ckiee, 2023-01-02:
I've set up [ckiee/logiops](https://github.com/ckiee/logiops) for now, spending a few hours on getting most of the smaller accumulated PRs in, [notes here](https://github.com/ckiee/logiops/blob/master/pixl-pr-todo.org).

I'm going to be emailing Pixl to see if they're still around on the nets and if not we can make my repo more permanent.

@chomosuke: Just [invited](https://github.com/ckiee/logiops/invitations) you, more people to split the potential future load would be nice. I'm pretty busy and I foresee I'll be even busier in a few months.

ckiee, 2023-01-02:
Update: It isn't abandoned!

Pixl says they're on a break and will come back eventually. Also they'll "look at getting additional maintainers for the project".

closed 2023-05-03 by PixlOne