PixlOne/logiops #349: Random behavioral error for gesture

VDuchauffour, 2022-12-31:
Hello, I use the following configuration for my MX Master 3 and MX Master 3S. I have mapped the gesture direction to my window manager keyboard shortcut. Everything works fine with the MX Master 3 but with the MX Master 3S I have some random errors. Between 1 in 5 and 1 in 10 times I get the opposite direction. Any idea where this problem is coming from?
Thank you
```````shell
devices: (
{
    name: "MX Master 3S";
    smartshift:
    {
        on: true;
        threshold: 10;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;   # gesture button
            action = 
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFTALT", "KEY_L"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFTALT", "KEY_H"];
                        };
                    },
                );
            };
        },
    );
}
);

chbornman, 2023-02-26:
Hi,
I would try adding a threshold value for each action. I have the 3S and it seemed to work for me. Just between the "type" and "keys" lines, add "threshold: 200;"

AndiB42, 2023-05-24:
Hi, 
i've the same issue on my MX Master 3S. It seems, that the mouse movement is recorded, even if the gesture button is not pressed yet. In my case it's reproducible like that: 
- move your mouse to the right first -> stop moving and press gesture button -> Action `Right` is triggered.
Same for any other direction. And even with a threshold configured. 
I've also tried the other modes `OnThreshold` with a threshold set and `OnInterval` with a interval set. Same behavior.

christophschw, 2023-05-25:
> Hi, i've the same issue on my MX Master 3S. It seems, that the mouse movement is recorded, even if the gesture button is not pressed yet. In my case it's reproducible like that:
> 
> --> move your mouse to the right first --> stop moving and press gesture button --> Action `Right` is triggered.
> 
> Same for any other direction. And even with a threshold configured. I've also tried the other modes `OnThreshold` with a set threshold and `OnInterval` with a set interval. Same behavior.

I can reproduce this issue. Setting threshold doesn't change it.

chbornman, 2023-06-02:
Thanks for the fix! Tried it out and it is working great!

AndiB42, 2023-06-02:
Thanks for informing! 
For me, the version 0.3.2 fixed this issue too! 
But.. unfortunately it's crashing more often now and i can't see any error logs.

VDuchauffour, 2023-06-15:
It seems that version 0.3.2 also solves the problem for me! I close the issue.