PixlOne/logiops #350: logi -v not able to detect MX Master 3s connected through USB receiver

larkrak, 2023-01-02:
Hi, everything was working properly till I went directly with USB connection and now Ubuntu cannot even detect the mouse.

This is the complete output for `logid -v`

[image]

Connected via bluettooth, the output was completely different. The mouse was detected and actually it told me the correct name of the devide to add it to the config file. 

Working on Ubuntu 22.04

Velho, 2023-01-04:
I think this was listed under the ArchLinux wiki, but I assume logiops does not support the new bolt usb receiver. The workaround is to use Bluetooth.

Tazerbot, 2023-01-21:
I also own this mouse and would love to help out in any way I can. Bluetooth works fine but the bolt adapter does not.

Velho, 2023-01-21:
@Tazerbot,

Have you tried out solaar? <https://pwr-solaar.github.io/Solaar/>
I found some success with it.

Tazerbot, 2023-01-27:
@Velho 

I have, yes, I'd just like to see more options (get it?) in the userspace. I like to help when I can.

macbeto, 2023-01-30:
I also has the same problem with my Logitech 3S and Bolt USB. I installed Solaar, but my thumb buttons (backward/forward) are not working. I don't know if it could be related to Linux Mint or the application.

cebms, 2023-02-28:
I'm experiencing a similar issue here. Solaar recognizes my 3S and is able to tune some settings, but the gesture button does not work. I was hoping I could use Logiops to make it work, but it does not recognize the mouse. Currently using Ubuntu 22.04

PixlOne, 2023-05-03:
Fixed in the latest release.

closed 2023-05-03 by PixlOne